UnchartedBull/OctoDash #1054: Display gcode thumbnail using Simplify 3D slicer  (S3D)

csabal, 2020-10-06:
Sorry for the question, but it is possible to display the gcode thumbnail using S3D? Or everybody using Cura / Prusaslicer. 

I tried to find a workaround, but did not find anything related here. 
Maybe I miss something......

UnchartedBull, 2020-10-15:
I think that isn't possible unfortunately. @jneilliii took care of writing the plugins for PrusaSlicer and Cura AFAIK, maybe he knows more. Closing this here for now, since OctoDash relies on a plugin to extract the image.

closed 2020-10-15 by UnchartedBull

NotExpectedYet, 2020-11-29:
> 
> 
> Sorry for the question, but it is possible to display the gcode thumbnail using S3D? Or everybody using Cura / Prusaslicer.
> 
> I tried to find a workaround, but did not find anything related here.
> Maybe I miss something......

https://github.com/NotExpectedYet/s3d-thumbnail-generator s3d thumbnails, only works on mac and linux couldn't figure out how to screenshot on windows.

boweeble, 2021-05-07:
I've just recently started using this software and wanted to use Simplify3D and have thumbnails so created this for windows.  Feel free to check it out here: https://github.com/boweeble/s3d-thumbnail-generator

Khaos66, 2022-02-07:
I created a feature request on the [S3D forum](https://forum.simplify3d.com/viewtopic.php?t=16267). It's still pending for review by a mod. But hopefully it will be visible to others soon.

Please voice your interest on this one, so they'll add it to S3D ;) 
Cheers

zelmil, 2022-05-17:
> Tried the windows version and it is not working.

IT works, but you need to insert:
c:\S3D_Thumbs\S3D_Thumbs.exe f="[output_file_path]"

instead of: c:\S3D_Thumbs\S3D_Thumbs.exe f=[output_file_path]

Critical are qoutation marks in: f="[output_file_path]"
(Of course change part: c:\S3D_Thumbs\ to your S3D_Thumbs.exe path if you put it elsewhere)

fabiorinaldus399, 2022-07-13:
I've written this code which uses openscad to create an image from the stl file imported in simplify 3d and then encode this image in the gcode. It has some limitations but I find it perfectly valid for most of the prints. It works fine with klipper.

https://github.com/fabiorinaldus399/gcode-tumbnail-generator-for-Simplify3D

jneilliii, 2022-07-13:
The slicer thumbnails plugin has links for other Simplify 3D solutions as well. https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails#simplify3d

fabiorinaldus399, 2022-07-13:
That's right, and I find the already existing script very useful in certain situation where you have multiple models loaded in simplify3d and you want to visualize them all (or other instances like that), the benefit of using openscad to create the image is that it doesn't rely on taking a screenshot of the application so you don't have to be careful about the position of the window, the resolution of the monitor, the position of the models and so on. Moreover as openscad is cross platform this method should teoretically still be compatible with linux, windows and MacOS

fabiorinaldus399, 2022-07-13:
Thank you, I'll try to add the linux version as soon as possible to increase compatibility

NotExpectedYet, 2022-07-13:
There's a second branch I never got around to putting on the master branch on my Linux version with openSCAD support already there. If I remember rightly it had some limitations due to file naming because s3d only gives you the output file path not the input path as a variable.

fabiorinaldus399, 2022-07-13:
Yes, that's true, but I've already addressed that in the repo where I wrote about the limitations of the code.
You can still run the code completely separate from simplify3d and manually select the gcode and stl file

fabiorinaldus399, 2022-07-13:
I've uploaded the linux version with the relative instructions

NotExpectedYet, 2022-07-14:
Ah that's a good way of going about it nice work

bolsoncerrado, 2022-11-14:
I've "installed" all the steps to generate S3D thumbnails on my MacOS computer.

Now my question is: is there any easy solution to batch-generate the thumbnails for all the old .gcode files rather than opening each, slicing, saving and closing? Or no workaround?

Thanks!

bolsoncerrado, 2022-11-14:
"by selecting the stl file"? What if a .factory file has multiple .STLs inside, like 99% of my .gcode library? hehe

bolsoncerrado, 2022-11-17:
FInal question: 

Any tips on how to "frame" the item being thumbnailed? I find out the snapshot taken for the thumbnail often shows mid air, mid print pictures instead of the item being printed.

Thanks!