UnchartedBull/OctoDash #1057: activate view mouse

asaku, 2020-10-08:
hello good to all I have a problem the touch of my screen is damaged and I wanted to put the mouse to see to handle the screen but I can not find how to put that function a greeting and thanks

UnchartedBull, 2020-10-09:
`nano ~/.bashrc` and remove the `-- -nocursor` part from this line: `xinit -- -nocursor`.

closed 2020-10-09 by UnchartedBull

slicerFPV, 2021-01-27:
I’m wondering how I ad mouse to mine aswell? What line can I put in the ssh to add the mouse for use on a computer monitor?

UnchartedBull, 2021-01-29:
Did you use ratpoison or a different desktop? Which user are you logged in while accessing the file (`whoami`). You need to be the same user as is used while showing OctoDash (pi mostlikely).

slicerFPV, 2021-01-30:
I believe I’m signed is as pi. I know how to sign in as su or as root but I didn’t install ass any of those user just pi. It did ask for my su password when I installed if I remember right. And yes it says welcome to rat poison when I boot up the pi. Iv attached a photo of the screen that comes up when I put in he command.

UnchartedBull, 2021-02-04:
Could you post the output of `ls -la`?