UnchartedBull/OctoDash #1062: touchscreen response on dev is slow

UnchartedBull, 2020-10-09:


UnchartedBull, 2020-10-11:
traced back to Electron 10 upgrade. Current master runs fine with Electron 9. Needs further investigation.

TheNeskik, 2020-10-22:
Could it be linked to Hardware Acceleration ?

I'll try this weekend to add
`app.disableHardwareAcceleration();`
to main.js

(Post this comment here as a remember for me)

UnchartedBull, 2020-10-22:
Can be, I just tried another dev build the other day and enabled the Fake KMS OpenGL driver for the Raspberry (which in fact enables GPU accelerated CSS). The performance was still sup-bar and probably a little worse than Electron 9 with no Hardware Acceleration. Electron 9 with hardware accelerated CSS was butter-smooth, so I think there is something else going on. But I never tried to disableHardwareAcceleration from Electron directly because it does give a notable performance improvement with the CSS animations.

UnchartedBull, 2020-10-22:
I found a few more things which could be the cause of this issue:

- **browserWindow.backgroundThrottling** might be something to look into (set to false)
- **browserWindow.webgl** Raspberry doesn't support WebGL out of the box, so disabling this might be worth a try as well
- try building with a higher version of node, this probably won't have any effect whatsoever but it is worth a try. I'm currently building everything with 12.16.1, while Electron 10 uses 12.16.3. So trying to build everything with 15 might be something to try.

Those three points probably won't do much, but I wanted to write them down somewhere and they might be worth a try.

closed 2021-03-18 by UnchartedBull

Vamoss, 2023-03-01:
> Could it be linked to Hardware Acceleration ?
> 
> I'll try this weekend to add `app.disableHardwareAcceleration();` to main.js
> 
> (Post this comment here as a remember for me)

This saved me hooooours of work trying to figure out why my application was so slow when running on touchscreen.
Thank you!

PR #1438 (2021-03-18): Switch to websockets
Not ready to merge yet. Sorry if you got all excited :/. This (currently) is just to keep track of what issues will be closed by this huge PR. If you have any thoughts - feel free to comment

Closes #1358 
Closes #1123 
Closes #595 
Closes #1177 
Closes #1374 
Closes #1505

Fixes #1304 
Fixes #1086 
Fixes #1062
Fixes #1447
Fixes #941
Fixes #1340
Fixes #1550