UnchartedBull/OctoDash #1071: Settings SAVE blanks screen and won't respond

M1dn1ghtN1nj4, 2020-10-12:
**Describe the bug**
When hitting SAVE in the settings menu, the screen goes blank and is unresponsive.

**To Reproduce**
Steps to reproduce the behavior:
1. Go into Settings
2. Hit SAVE
3. Witness bug

**Expected behavior**
Should return to the main screen

**General Information:**
 - Raspi 4B
 - OctoPi: 0.18.0
 - OctoPrint Version: 1.4.2
 - OctoDash Version: Build as of 10-11-2020

komandrik, 2020-10-21:
I confirm. I get the same problem after updating to version 2.0.1 of Raspberry Pi 3B +

UnchartedBull, 2020-10-21:
Will release V2.1.1 soonish to fix this.

closed 2020-10-23 by UnchartedBull

M1dn1ghtN1nj4, 2021-01-31:
> Closed #1071 <https://github.com/UnchartedBull/OctoDash/issues/1071> via

I'm having a lot of issues recently with DisplayLayerProgress/values not
being recognized by OctoDash.  404 errors.  Now, I'm also unable to get
DisplayLayerProgress to actually install.  I'm not entirely sure what's
going on here.  Still working on a few things, but I feel like I'm running
in circles.  Has that plugin been giving you any grief?  Or is it just me?

UnchartedBull, 2021-01-31:
Lots of issues piling up regarding DLP atm again. Working on switching over to websockets, this way DLP ignores can just be ignored. Olli did publish an update 1.15.2 which is supposed to fix some issues though.

PR #1079 (2020-10-14): Don't blank after saving settings
Fixes #1071.
PR #1116 (2020-10-23): saving works again + packaging improvements
Fixes #1071.