UnchartedBull/OctoDash #1090: Octodash setup error "Http failure response for http://localhost:5000/api/connection: 403 FORBIDDEN"

Labdoc1957, 2020-10-18:
**What doesn't work?**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**What did you already try?**
A clear and concise description of what you tried to make OctoDash work.

**General Information:**
 - Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]
 - OS Info [e.g. Raspbian Buster Lite, please also indicate if you used OctoPi]
 - OctoDash Version [e.g. v1.0.0]
 - OctoPrint Version [e.g. v1.0.0]

**Additional context**
Add any other context or screenshots about the feature request here.

Labdoc1957, 2020-10-18:
Octodash setup error "Http failure response for http://localhost:5000/api/connection: 403 FORBIDDEN"

So I set up Octoprint on my Pi4B 4Gb and on my Pi4B 8Gb which are working as expected.  I'm trying to set up a BTT PiTFT50 on my 4Gb Pi.  I went through the install process and the OctoDash starts up automatically, however, I'm unable to get past the setup screens.  I go through all of the screens and the whole thing comes to a crashing halt on the validation (final) screen. Any thoughts on how to get past this?
[image]

UnchartedBull, 2020-10-19:
Seems like your token is wrong. V2.1.0 (which should be released later today) can automatically generate your token (you just need to confirm it on the OctoPrint UI). Please wait for the release and try again.

closed 2020-10-26 by UnchartedBull