UnchartedBull/OctoDash #1094: Failed to fetch http://ftp.halifax.rwth-aachen.de/raspbian/raspbian/pool/main/w/wayland/libwayland-server0_1.16.0-1_armhf.deb

Remklep, 2020-10-19:
Getting the following error during the automatic installation:
Failed to fetch http://ftp.halifax.rwth-aachen.de/raspbian/raspbian/pool/main/w/wayland/libwayland-server0_1.16.0-1_armhf.deb

Remklep, 2020-10-19:
My appologies, this is using a fresh install of the OctoPrint 0.17.0 image on a Pi 4B 2Gb.

FrankX0, 2020-10-20:
Can you try again? I can fetch it now.
There seemed to have been some issues with this server (#1091 ).

UnchartedBull, 2020-10-20:
I don't have any access to these servers, so trying again in an hour or two is always a good idea. If that still doesn't work installing the package manually (with a different mirror) should remove that error. Since other users reported the same problem (as @FrankX0 mentioned) and for them it's fixed now, I'm closing this here.

closed 2020-10-20 by UnchartedBull