UnchartedBull/OctoDash #1096: Possible to increase vertical scrollbar width?

CaptainSweatpants, 2020-10-19:
Is there any way to increase the width of the scrollbar; or better yet, make the entire screen "scrollable" via touch? Its just a tad tricky to get my big fingers on the very small scrollbar on the right side.

UnchartedBull, 2020-10-20:
Normally the full screen is scrollable via touch. What kind of display are you using?

CaptainSweatpants, 2020-10-21:
LCD-SHOW 4".
Yeah i can only scroll with the tiny horizontal scrollbar,

UnchartedBull, 2020-10-22:
It's one that is connected via GPIO only, right? Have you tried using the waveshare driver (https://github.com/waveshare/LCD-show), I don't know whether this fixes your problem, but it may be worth a try. Can also be a limitation of your hardware, as these cheap displays sometimes don't have enough touch resolution to track a "swipe" and rather send multiple click events.

There is unfortunately nothing I can do to get fullscreen scrolling working with these displays. And increasing the width of the scrollbar would mean cramping the other information even more and it would also make the scrollbar obnoxiously big on 5" and 7".

thegarbz, 2020-10-24:
If it is in fact just a big finger problem and not a touchscreen problem you can customise the scrollbar width with a custom theme. 
Since I'm playing with a themes at the moment I may be able to help you with this, but as I'm commuting to Germany right now this may take a couple of weekends for me to get to.

UnchartedBull, 2020-10-26:
Thanks to @thegarbz the bigfinger theme is available (Thanks for the PR)! Please try it with your installation and check whether scrolling is now a little more convenient :)

PR #1130 (2020-10-26): New Theme BigFingers by theGarbz
This theme is a minor variation on the default theme with extra wide scrollbars to suit smaller displays where the scrollbar is sometimes hard to use. 

Resolves #1096
closed 2020-10-26 by UnchartedBull