UnchartedBull/OctoDash #1097: Can't get OctoPrint script...

darouss99, 2020-10-20:
Hi, 

I just updated from OctoDash 2.0.0 to 2.1.0.  The update seem to go well, but after rebooting, I have the message "Can't get OctoPrint script! Please restart your machine.  If the error persists open a new issue on GitHub.".  After a new reboot, the problem remain.

**What did you already try?**
I tried to detele the file ~/.config/octodash/config.json and reboot. The wizard start, and It's working well.  But when I reboot again I have the initial error message.

I tried to uninstall the 2.1.0 and install back the 2.0.0, It's working well. But when I install again the 2.1.0 the problem come again.

**General Information:**
 - Raspberry PI 3B+
 - Raspbian Buster (not used OctoPi])
 - OctoDash Version v2.1.0
 - OctoPrint Version v1.4.2

Do you have an idea of the problem occur on my installation?

Thank you for your help

UnchartedBull, 2020-10-20:
This needs a bit of an explanation. I hoped that this error will not occur, but that took only 6 hours so yeah ...

With the next minor release OctoDash will be switching over to the OctoPrint JS library. This comes with many improvements (faster updates, less load, no DLP required, ...). The problem is, that the library isn't available to download on npm where all the other packages are available, but rather needs to be pulled of the OctoPrint instance.

Here is the problem: OctoDash definitely 100% needs this script (from v2.2.0 onwards) to do anything. This means that before the main-screen could be shown the script needs to be downloaded from the OctoPrint instance, which in fact means that OctoPrint needs to be fully started before OctoDash could start. This can create a race condition, which could lead to this error.
v2.0.0 and below didn't use the script, so they just suppressed any error for the first 30 seconds or so until OctoPrint was available. This won't work for future versions anymore though.

TL;DR OctoDash needs to wait until OctoPrint is fully started, which could take longer on some systems than the current of around 15 seconds.

If you also encounter the problem please comment here with the following information:

- Raspberry Pi Model
- OctoPrint (and Python) version
- Do you delay the start of OctoPrint during boot?

I'll try and provide a fixed version of OctoDash with a more rigid handling of the script loading in the next days (link attached to this issue). Until then please downgrade to v2.0.0.

darouss99, 2020-10-20:
Thank you for the explanation. Is there a way to delay the start of Octodash?

Raspberry Pi Model : 3B+
OctoPrint (and Python) version : 1.4.2 - 2.7.16 and 3.7.3
Do you delay the start of OctoPrint during boot? Yes

UnchartedBull, 2020-10-20:
By how long are you delaying the start? Best way to delay the OctoDash start is probably by adding a `sleep <seconds>` to `~/.xinitrc`.

eliteSchwein, 2020-10-20:
You can start octoprint via a service file, i can sent you my solutions when you want it :)

darouss99, 2020-10-20:
I tried the @UnchartedBull method. 20 sec with the sleep command.  It's working great. 10 seconds wasn't enough.  I did not tried the @eliteSchwein solution, I don't know wich one is best.

I have another symptom, I don't know if it's related.  But when I save the config, the screen goes black, I need the restart. The config is saved correctly.

Thank you for your help.

eliteSchwein, 2020-10-21:
#1071

ax42, 2020-10-21:
Same issue.  Running on a mini-PC (celeron), starting via a user service file.  Octoprint runs in a docker container for me (the official one), so I can't wait for the octoprint service.  Added 'sleep 15' to my start script which helps it work.

Perhaps octodash could cache the file it needs and only download new versions when it knows octoprint is running?

UnchartedBull, 2020-10-21:
Really don't wanna open that caching box. I mean there is no useful information to show if OctoPrint is unreachable, so my plan for this is to wait for the port defined in the config to be open and then query the script (with a timeout of a minute or so). Currently testing a few things with that.

whoim2, 2020-10-22:
pi3b (non plus)
OctoPrint 1.4.2 Python 2.7.16 OctoPi 0.17.0
no
have error

DarKOrange75, 2020-10-28:
> Well my setup is up and online with a sd card in 15s or so

Hi, can you please give more details about your solution? I had to delay octodash start by 60 secondes :-(
Thank you!

DarKOrange75, 2020-10-28:
@UnchartedBull 

Hi, OctoDash on my setup can't find the Octoprint script unless I manually refresh the Octoprint Web interface in a browser... 
As explained above I have delayed Octodash start by 60 seconds in xinitrc, I went up to 120 seconds but Octodash wasn't finding the script unless I had reloaded the web interface.

eliteSchwein, 2020-10-28:
Make a new issue

UnchartedBull, 2020-10-29:
@DarKOrange75 does OctoDash automatically connect to your printer? I currently have an idea on how this can be improved. I'll probably update v2.1.1 thenn. I'll let you know in this issue.

DarKOrange75, 2020-10-29:
> @DarKOrange75 does OctoDash automatically connect to your printer? I currently have an idea on how this can be improved. I'll probably update v2.1.1 thenn. I'll let you know in this issue.

Yes it does. Thank you!

whoim2, 2020-10-29:
Try 2.1.1 now, not connect, error as above

LouDnl, 2020-11-03:
RaspBerry Pi 3B
OctoPrint version: 1.4.2
OctoDash version: 2.1.1
Python 2.7.16 
Python 3.7.3 

No delay on OctoPrint or OctoDash

Will add a delay on OctoDash startup to try if this works.

jaymcd, 2020-11-03:
> Well my setup is up and online with a sd card in 15s or so

EliteSchwein- can you share your setup?

UnchartedBull, 2020-11-10:
Quick update: v2.1.2 will now try forever to load the script (first 30 seconds each second, after that every 15 seconds). There is also a hint shown after 30 seconds to check your printer connection. If you experience any issues after upgrading to v2.1.2 (once it's released). Please comment in this issue. Thanks.

closed 2020-11-10 by UnchartedBull

DarKOrange75, 2020-11-14:
When do you plan to release 2.1.2 @UnchartedBull ?

UnchartedBull, 2020-11-14:
I just found out the other day that I can do everything without the OctoPrint script. So release v2.1.2 will actually have the Script loading removed altogether (which should greatly improve startup performance). This takes a little more time though. Will be released early next week hopefully.

DarKOrange75, 2020-11-14:
Ok thank you @UnchartedBull for answering so fast. I'll wait.

binsweden, 2020-11-15:
how can i downgrade back to version 2,0,0 when im having these problems??

Buddinski88, 2020-11-15:
@binsweden, would be good to know because i got the same issue :-(

binsweden, 2020-11-15:
> 
> 
> @binsweden, would be good to know because i got the same issue :-(

found this and it worked

Buddinski88, 2020-11-15:
> > @binsweden, would be good to know because i got the same issue :-(
> 
> found this and it worked
> 
> ```
> Download the latest release. Check for newer version (Releases) and replace v2.0.0 with the latest version
> wget -O octodash.deb https://github.com/UnchartedBull/OctoDash/releases/download/v2.0.0/octodash_2.0.0_armv7l.deb
> Install the app
> sudo dpkg -i octodash.deb
> Reboot (or Restart OctoDash)
> ```

Thanks, mate. That worked! :-)

PR #1117 (2020-10-23): Better connection handling
Closes #1097.
PR #1167 (2020-11-10): improved script loading
Fixes #1097.