UnchartedBull/OctoDash #1099: No startup Icon

eliteSchwein, 2020-10-20:
**Describe the bug**
Startup is just grey.

**To Reproduce**
Steps to reproduce the behavior:
Start the Programm

**Expected behavior**
Old startup logo is visible while octodash starts

**Screenshots**
 -tomorrow-

**General Information:**
 - Hardware waveshare 7", rpi4 4gb
 - OS Info raspbian buster 32bit desktop, i manuelly installed octoprint
 - OctoDash Version 2.1.0
 - OctoPrint Version 1.4.2

**Additional context**
You are amazing!

UnchartedBull, 2020-10-22:
I don't seem to be able to replicate this, neither on the Pi nor the dev app. Could you maybe check if the icon does appear for a split-second during startup and disappears afterwards?

eliteSchwein, 2020-10-22:
its just blank grey screen

eliteSchwein, 2020-10-23:
ok switching the cable fixed it

closed 2020-10-23 by eliteSchwein