UnchartedBull/OctoDash #1100: Turn off backlight when sleeping, not just display

tomaae, 2020-10-20:
**Is your feature request related to a problem? Please describe.**
Option "Turn Screen Off While Sleeping" turns off display, but not backlight.

**Describe the solution you'd like**
Since there are many types of displays which are handled differently, I propose following solution:
Add 2 more options to config:
1. Execute command in cli when screen is turned off
2. Execute command in cli when screen is turned on

For example in my case it would be:
1. `gpio -g pwm 19 0`
2. `gpio -g pwm 19 40`

UnchartedBull, 2020-10-21:
I added this to #410 as all the screensaver improvements will be tracked there.

closed 2020-10-21 by UnchartedBull

whoim2, 2020-10-29:
I solved this by connecting the backlight to the power-on pin (plug-in for octoprint) in parallel with the power supply relay, to an unused pair of contacts

tomaae, 2020-10-30:
Thats not an option with my display unfortunately. I use full header display and messing with that would void warranty. I rather avoid that since its not cheap display.