UnchartedBull/OctoDash #1103: SAVE hangs octodash

Protoncek, 2020-10-21:
**Describe the bug**
After upgrade to v2.1.0 hitting "save "button crashed octodash (just goes black) and restart of RPi is required

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Settings
2. Click on 'Save'
3. that's it...

Is it just mine installation or...? I updated through octodash interface. When octodash freezes octoprint server works normally, so i can restart RPI from octoprint's web site.

UnchartedBull, 2020-10-21:
Duplicate #1071.

closed 2020-10-21 by UnchartedBull

Protoncek, 2020-10-21:
Ups... sorry, i didn’t look so far back, since v2.1.0 was released only 2 days ago and it didn’t happen to me in v2.0.0... so i automatically thought that it’s new version’s fault.
My bad...