UnchartedBull/OctoDash #1104: Error while updating to 2.1 either via touchscreen or from command line

skibum-za, 2020-10-21:
I tried to update to octodash 2.1 from 2.0 and it failed due to a missing update-octodash file. So I fixed that and ran the upgrade again (from the touchscreen). This then gave another error. So I tried from the command line and it reported the same error :- 

` wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/update.sh | bash`
`Updating OctoDash`
`octodash.deb                     100%[=======================================================>] ` `62.11M  7.92MB/s    in 8.0s`
`(Reading database ... 57993 files and directories currently installed.)`
`Preparing to unpack octodash.deb ...`
`Unpacking octodash (2.1.0) over (2.1.0) ...`
`Setting up octodash (2.1.0) ...`
`/var/lib/dpkg/info/octodash.postinst: line 10: update-desktop-database: command not found`
`Processing triggers for hicolor-icon-theme (0.17-2) ...`
`Processing triggers for mime-support (3.62) ...`
`Done. Restart your Raspberry Pi to start the newest version!`

As you can see tits the 'update-desktop-database' command not found error. 

It still reports that I'm on 2.0 from the touchscreen. 

I then tried doing the 'manual' upgrade after downloading the octodash.deb file and again that fails with the same error :- 

`sudo dpkg -i octodash.deb`
`(Reading database ... 57993 files and directories currently installed.)`
`Preparing to unpack octodash.deb ...`
`Unpacking octodash (2.1.0) over (2.1.0) ...`
`Setting up octodash (2.1.0) ...`
`/var/lib/dpkg/info/octodash.postinst: line 10: update-desktop-database: command not found`
`Processing triggers for hicolor-icon-theme (0.17-2) ...`
`Processing triggers for mime-support (3.62) ...`

UnchartedBull, 2020-10-21:
The error shouldn't influence the upgrade. If you want to get rid of it install `desktop-file-utils` (has been added to the installation script only since about a month or so). The really weird thing is that the console is reporting the correct version (2.1.0), so the version is definitely right there. Could you maybe check your settings and see whether you see these 3 options und "octodash"?

[image]

spiff72, 2020-10-23:
I was getting this same error when updating from the UI.  I just installed 2.1.0 a few days ago, and ended up using the "manual" update script from the command line with ssh.

I just saw the notice on the settings icon showing the update was available, so tried to update via the UI again, and this error occurred.  After seeing this issue report, I tried running sudo apt update and sudo apt install desktop-file-utils, and then tried to update again via the UI.  This time the progress bar that displays while the updating is taking place appeared to freeze for a little while, but eventually I got the message asking whether to reboot to complete install (and did).  

This update appeared to work after installing the desktop-file-utils.  Not sure why it didn't get installed after I did the manual install of 2.1.0 though...

spiff72, 2020-10-23:
The interface definitely seems more responsive and less laggy after today's update!

Shadow27374, 2020-10-23:
> The interface definitely seems more responsive and less laggy after today's update!

Yes it has become better. But a little more would be nice.

UnchartedBull, 2020-10-25:
There is only so much CPU rendered interfaces can do. This currently is the limit (at least to my knowledge). I'm currently testing hardware acceleration (which does enable super smooth animations) but does require some setup. If this all works out and works I'll maybe add an entry to the wiki on how to enable that.

Shadow27374, 2020-10-25:
The previous version did not have this problem or was relatively fast. After the first update it was terribly slow and after the last update it is almost okay.

I had to start the updates via console. Üor the display it stops at some point.

UnchartedBull, 2020-10-26:
@Shadow27374 have you installed the update script as described here: https://github.com/UnchartedBull/OctoDash/wiki/Update#in-app-update?

Shadow27374, 2020-10-26:
I didn't do that, that explains that of course.

UnchartedBull, 2020-10-26:
Was there an error message that popped up?

TobsBobs, 2020-10-26:
I have the same issue too.

wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/update.sh | bash
/var/lib/dpkg/info/octodash.postinst: line 10: update-desktop-database: command not found

UnchartedBull, 2020-10-26:
Then the same fix applies to you. Install `desktop-file-utils`

TobsBobs, 2020-10-31:
@UnchartedBull thanks, my problem ist solved.

closed 2020-11-02 by UnchartedBull