UnchartedBull/OctoDash #1107: config invalid -> octodash should have required property 'previewProgressCircle'

Sekisback, 2020-10-22:
**Describe the bug**
after updating from 2.0 to 2.1 i got this error message.
config invalid -> octodash should have required property 'previewProgressCircle'

**To Reproduce**
Steps to reproduce the behavior:
update via display error occurred
downgrade via comandline
update via comandline error occurred

**General Information:**
 - RPi 3B
 - OctoPi 0.17.0
 - Octoprint 1.4.2 without python3
 - OctoDash 2.1
 

**Additional context**
downgrade back to 2.0 everything work like expected

TheNeskik, 2020-10-22:
Hi, you can look to the wiki, this is a common error when updating, this is the right part of the wiki :

> Config is invalid
Problem
You're getting a message, telling you that your config is invalid alongside with some weird technical-looking things. This can either happen if you upgrade to many versions at a time (I'm trying to merge configs from the previous release, if possible, so if you're upgrading from 1.4.0 to 1.4.1 you should be fine, but if you're going directly to 1.4.2 you might run into this error)

> Fix
If you're unsure about what the errors mean you can always just delete that file rm ~/.config/octodash/config.json and reboot your Pi. This will open the initial setup screen again and will create a new, valid config for you. All your configuration data for OctoDash will be lost, OctoPrints data isn't touched.

 > If you understand what needs to be changed in your config please go ahead and edit it with your favorite text editor. You can find the config at ~/.config/octodash/config.json.

> If you want more information about the config / settings have a look here.

so through SSH : you can `sudo rm ~/.config/octodash/config.json` then `sudo reboot`
it will reset the octodash config but it is really fast to configure so have fun :)

UnchartedBull, 2020-10-22:
did you install an intermediate nightly build? That's why the automatic config fix probably didn't kick in. Either add `previewProgressCircle: false` to your config (under `octodash`) or remove the config and start from scratch (it's way easier to setup with v2.1.0!).

closed 2020-10-22 by UnchartedBull

Sekisback, 2020-10-22:
> did you install an intermediate nightly build? That's why the automatic config fix probably didn't kick in. 

yes I tried a nightly build a few days before.
I removed the config.json and installed octodash manual, now everything works like expected. thanks for your support