UnchartedBull/OctoDash #1108: OctoDash Custom Setting  change order of items listed

3dattic, 2020-10-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. If applicable also link other issues / bugs here.

no problem, this is a feature change request. The current order of the Custom setting is to have them with the GCODE/Action first then, Icon setting and Color setting with the icon picture displayed on the same line. below that is two additional boxes to select.

the better order is to have the Icon and Color first with the icon picture and below that is the Gcode/Action and finally the two selction boxes. This makes it easier to follow and shows where that section finishes before going to the next item.

Second item have the menu full screen so it's easier to read and navigate as the scroll bar is small and text is small

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Change the order of the boxes in the custom setting to make it clearer to the end user
make the menu boxes full screen making it more readable

**Link other projects**
If you want OctoDash to include other OctoPrint plugins or similar please link them here.

**Additional context**
Add any other context or screenshots about the feature request here.

UnchartedBull, 2020-10-26:
There is already an issue open regarding text readability on smaller screens (for now please use the big fingers theme).

Regarding the order of the items on screen: I'll have a look at that and check how it looks. Not 100% sure whether this makes it into the final release since I find it easier to understand the way it currently is.

UnchartedBull, 2020-11-17:
Decided to leave it the way it is.

closed 2020-11-17 by UnchartedBull