UnchartedBull/OctoDash #1113: Laggy Ui (Hyperpixel 4.0) 3b+

jjuuzzii, 2020-10-22:
I‘be got good ammunition in fex Files but the settings do lag very much. It’s a non fluid movement. 
If I want to change temps in Print mode it’s also very laggy/slow

TheNeskik, 2020-10-22:
What version of Octodash do you use ?
Latest version (2.1.0) doesn't any issue on my Pi3

jjuuzzii, 2020-10-22:
Yea 2.1

jjuuzzii, 2020-10-22:
Is you zu completely fluid ? 
Whic screen ?

UnchartedBull, 2020-10-22:
This could be due to the new background-blurring effect introduced in v2.1.0. I noticed this on my 3B as well in some cases. There is a fix for that, although not fully mature yet. I think I'll remove the blurring for v2.1.1 and then reintroduce it later on.

jjuuzzii, 2020-10-22:
Ok thanks!

UnchartedBull, 2020-10-22:
v2.1.1 hopefully fixes that, if not please reopen the issue.

sergejvoronov, 2020-10-22:
I can confirm the issue just after OctoDash update from 2.0 to version 2.1

closed 2020-10-22 by UnchartedBull

jjuuzzii, 2020-10-22:
How can I update to 2.1.1?

UnchartedBull, 2020-10-23:
Either use nightly builds linked in the README (although they haven't been updated for some time). Or wait until the official release of v2.1.1 (hopefully later today). You can always build it yourself from master if you prefer that.

Krontx, 2020-12-02:
I am a total newbie... How do I apply this fix: "remove backdrop filter and add dot animation #1114", my UI is very laggy.

UnchartedBull, 2020-12-04:
The fix is automatically applied. The display you are using is connected via the GPIO headers, right? If so the display just doesn't support a refresh rate of more than 2Hz, nothing I can do about this on OctoDash side.

Krontx, 2020-12-04:
@UnchartedBull I would happily switch the screen if that'd improve my experience. What screen are you using, what's the minimum recommended refresh rate?

UnchartedBull, 2020-12-04:
Probably worth trying out https://www.testufo.com/ in a browser and check what framerate is reported there. I'm running the official 7" screen which does have 60Hz, but 30Hz should work as well.

GPU isn't supported right now since it does create more problems and the animations are plenty smooth with CPU animations only.

You can head over to the releases and download any version you want (with `wget`) and then just `sudo dpkg -i <file>`. That should install the downloaded verison.

Krontx, 2020-12-04:
I fixed the issue, after realizing the problem wasn't the Pi 4 Model B performance itself, I jumped to troubleshoot the screen, it was easier than I though.

 All I had to do was:
1. Overclock the processor a bit (This step might not be completely necessary but why not if you have a fan on your Pi.)
2. Update the refresh rate from the stock 20MHZ to 62MHZ
3. Update the FPS from 20 to 60

Anybody using Adafruit 3.5" and a Raspberry Pi 4 Model B please note how to fix the slow/laggy UI, bad performance issuue.

```
1. Sudo nano  /boot/config.tx
2. Scroll down (Using the arrows) until "#uncomment to overclock the arm. 700MHZ is the default" left the value at arm_freq=800
3. Continue to scroll down to "dtoverlay" line that is towards the very end of the file and switch the value there with the following:` dtoverlay=pitft35-resistive,rotate=90,speed=60000000,fps=60 ` 

```

UnchartedBull, 2020-12-08:
Overclocking your screen is definitely next level 👍 Glad you figured it out!

caesarius, 2021-10-02:
So, can I assume that my issue with unresponsive/laggy/random performance is because OctoDash needs more processing power than my RPi 3B and Adafruit PiTFT 3.5" can provide?

My RPi is connected to a Prusa MK2S. Touch UI feedback is shown on screen (tap animation), but it triggers the intended action just randomly or not at all (I can upload a video if that's useful to illustrate this behavior).

I've found out about Octodash and _really_ like it (over TouchUI) mainly because its usability and simplicity, but I'd like to know if my current setup is not enough, hence I need to get a better hardware setup in order to make it work as intended.

If so, just upgrading the screen should be enough? Or do you suggest to upgrade the RPi as well. Thanks!

PR #1114 (2020-10-22): remove backdrop filter and add dot animation
Fixes #1113.