UnchartedBull/OctoDash #1118: Settings are hard, if not impossible, to open

Syco54645, 2020-10-24:
**Describe the bug**
The gear in the upper right corner is very hard to hit, most of the time I end up triggering the right most button in the center.

**General Information:**
rpi 4 on a 7 inch screen (AT070TN92) with a capacitive touchscreen on top. The hardware registers as an egalax. In octopi the screen works as does the touch but the touch is rotated 90 degrees. I solved this by adding the line `Option "TransformationMatrix" "0 -1 1 1 0 0 0 0 1"` to the touchscreen input class in the file `/usr/share/X11/xorg.conf.d/40-libinput.conf`

**Additional context**
I have calibrated the touch panel more times than I can count. Generally I do this by issuing `sudo DISPLAY=:0.0 xinput_calibrator --misclick 0`

dani190, 2020-10-25:
Noticed this also but thought it was just me... I actually have never been able to open the settings menu. Screen seems to be accurate touch wise though.

UnchartedBull, 2020-10-25:
I'll try and increase the touch box of the settings menu with the next release :)

M1dn1ghtN1nj4, 2020-10-25:
I haven't noticed a problem with it.  The only thing that seemed "off" to me, is the vertical scroll bars.  You have to hit just to the right of them to actually engage with them.  Touching directly on them actually counts as a touch inside the scrolling window.  Everywhere else in the system seems to have it's touch box where it should be.  Some do seem to be a bit larger than they need to be, but they still work.  I'll check this again tomorrow.

thegarbz, 2020-11-07:
@Syco54645 @dani190 @M1dn1ghtN1nj4 
Try this theme: https://pastebin.com/qKLEMzP0 (simply save it as "~/.config/octodash/custom-styles.css")
Your issue with not being able to hit the settings directly is almost certainly a problem with your calibration. The entire settings icon as well as the visible component of the scroll bar are active areas. 
This theme is a slight variation on my "BigFingers" theme which I made to address another user's issue with scroll bars and smaller screens. It gives much wider scroll bars but I've modified it to also increase the active area padding around the settings icon so it should be much easier to hit. Previously the active area ended flush with the bottom of the icon, now it extends down and to either side significantly including moving it out from the edge slightly. 

Please let me know if this works, if so I'll update my custom theme here in the repo with the larger padding too. 

[image]

@UnchartedBull When I hear back from them if this fixes it I'll ping you a PR to update the BigFingers theme.

dani190, 2020-11-08:
@thegarbz Just tried this and no luck. I am almost 100% sure it's not calibration as if I go to the control page, I am able to hit the question mark in the top right with no issues.

thegarbz, 2020-11-08:
@dani190 Not sure what the issue is then since I can't reproduce, not in the dev environment nor in the raspberry pi. Here's a demonstration of the above css file on a 7" screen https://youtu.be/F3OQWPKUY4U the hitbox is massive and above all active all over including in the gear icon itself. 

Is there some kind of drawing / testing utility you can use to verify you're actually getting the correct touch response in that part of the screen?

@UnchartedBull I'm going to put a PR in for the change anyway since it makes sense for the Big Fingers theme. I'll come with a general theme update this afternoon with a few tweaks to my other themes and a new one.

UnchartedBull, 2020-11-10:
Increased touchbox size is now also in the main theme from OctoDash. 

Can't adjust the touchbox size of the scrollbar though, but for me they're plenty big and work no matter where I touch them. If it's on the scrollbar thumb scrolling is working correctly via scrollbar.

PR #1168 (2020-11-10): increase settings touchbox size
Fixes #1118
closed 2020-11-10 by UnchartedBull