UnchartedBull/OctoDash #1123: Add Enclosure plugin Outputs as configurable actions

haplm, 2020-10-24:
**Is your feature request related to a problem? Please describe.**
I would like to turn LED on my enclosure ON and OFF with OctoDash. However, I'm not using Neopixel, but 12V LEDs controlled by a relay attached to one of the RPi GPIO pins. That pin is configured in Enclosure plugin as output (http://octopi.local/plugin/enclosure/outputs) like this:
```[{"index_id": 1, "label": "LED"}]```

**Describe the solution you'd like**
Include Enclosure outputs as predefine actions (https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions#predefined-actions)

Thanks!

jbuszkie, 2020-10-29:
Is there a way to do this (work around) until V3 comes out?

UnchartedBull, 2020-10-29:
I don't know of any. I'm not that familiar with Enclosure Plugin. Best guess would be to check whether you can define custom GCodes and just send those from OctoDash.

Zapfmeister, 2021-02-16:
Enclosure seems to provide an API to trigger outputs:
https://github.com/vitormhenrique/OctoPrint-Enclosure/wiki/API-Control

You can request all current gpio status with
'''
http://**{URL}**/plugin/enclosure/getOutputStatus?apikey=**{APIKEY}**
'''

Regular GPIO
'''
http://**{URL}**/plugin/enclosure/setIO?status=**{STATUS}**&index_id=**{OUTPUT_ID}**&apikey=**{APIKEY}**
'''
OUTPUT_ID is the automatic generated number id from octoprint-enclosure setting
STATUS is can be true to turn it on and false to turn the output off

UnchartedBull, 2021-02-17:
I actually wrote that API for the enclosure plugin (mainly to retrieve the temperatures). Should be fairly easy to achieve, it will just be a dumb toggle though, since the custom actions aren't able to handle states.

PR #1438 (2021-03-18): Switch to websockets
Not ready to merge yet. Sorry if you got all excited :/. This (currently) is just to keep track of what issues will be closed by this huge PR. If you have any thoughts - feel free to comment

Closes #1358 
Closes #1123 
Closes #595 
Closes #1177 
Closes #1374 
Closes #1505

Fixes #1304 
Fixes #1086 
Fixes #1062
Fixes #1447
Fixes #941
Fixes #1340
Fixes #1550
closed 2021-03-18 by UnchartedBull