UnchartedBull/OctoDash #1126: Update from 2.0 to 2.1.1 from UI hanging

morphias2004, 2020-10-25:
Running Octodash v2.0. on a Pi4 4Gb with Octoprint 1.4.2, OctoPi 0.17.0 and Python 3.7.3 with a HyperPixel 4.0.

UI notified of an update to 2.1.1 was available. I confirmed the update and it went to a progress bar that says "Installing update" and "This may take a while".

20mins later and it's still sitting there with this message. :(

morphias2004, 2020-10-25:
I've worked around this by hard power cycling the Pi and running the update script from the Pi command line.

thegarbz, 2020-10-25:
@morphias2004 This is addressed in the wiki and the release notes for 2.1.0. You need to add a script to get updates working, in the future. https://github.com/UnchartedBull/OctoDash/wiki/Update#in-app-update

@UnchartedBull given these releases came so close to each other it may be worth putting it in the release notes for 2.1.1 as well.

ruralaccent, 2020-10-25:
My in app update isn't hanging, but it's taking almost 50 minutes to download. When I update my other Octoprint on another Pi via Putty, it takes only a couple of minutes.

carl1961, 2020-10-25:
Update was same as @morphias2004 , doing wiki failed, because I had no script folder. I got all that fixed and got the Octodash to update, but update seems broke.  " Can't get Octoprint script!"  this is how I got the update script installed and working.

**Making Octodash update from OctoDash**

I did not have a scripts folder so i had to make one

Log in to pi  with putty (or whatever terminal you choose)

  ******   Creating  scripts  directory  ******   

  mkdir scripts 

***** creating script  ~/scripts/update-octodash   ******

sudo nano ~/scripts/update-octodash

**** add this  ****

#!/bin/bash

dpkg -i /tmp/octodash.deb
rm /tmp/octodash.deb

****save and exit nano  Ctrl o  Ctrl x ******

*** Make the script executable:

sudo chmod +x ~/scripts/update-octodash

****** Enable passwordless sudo for the update script (/etc/sudoers.d/update-octodash): ****
**** I did not have file , so this is new file too*****

sudo nano /etc/sudoers.d/update-octodash

****** add this ****

pi ALL=NOPASSWD: /home/pi/scripts/update-octodash

****save and exit nano  Ctrl o enter Ctrl x ******

***** Then reboot pi *******

sudo reboot

[image]
[image]
[image]
[image]
[image]
[image]
[image]

carl1961, 2020-10-25:
@ruralaccent  that would be most like your wifi speed, mine download less that a minute

carl1961, 2020-10-25:
I even tried adding the delay as once mentioned before , but still not working, " Can't get octoPrint script"
delay worked, as i could see the delay after ratpoison 
********************************
Edit:  UnchartedBull

By how long are you delaying the start? 
Best way to delay the OctoDash start is probably by adding a sleep <seconds> to ~/cd..
.
********************************
***********login as root and not pi ********************

nano /home/pi/.xinitrc

***** add sleep 20 above octodash **********
sleep 20

****save and exit nano  Ctrl o enter Ctrl x ******

***** Then reboot pi *******

sudo reboot

carl1961, 2020-10-25:
even manually updating from Unofficial OctoDash Builds  still "can't get OctoPrint script!"

[image]
[image]
[image]

UnchartedBull, 2020-10-25:
Is your printer connected to OctoPrint?

carl1961, 2020-10-25:
**BackDated to 2.0.0 and running again**
removing octodash

sudo apt-get  remove octodash
*********************************
**Download the latest release Check for newer version and replace v2.0.0 with the latest version (Releases)**
*********************************
wget -O octodash.deb https://github.com/UnchartedBull/OctoDash/releases/download/v2.0.0/octodash_2.0.0_armv7l.deb
*********************************
**Install the app**
*********************************
sudo dpkg -i octodash.deb

[image]
[image]
********************************
 Then reboot pi 
********************************
sudo reboot

carl1961, 2020-10-25:
@UnchartedBull  if your referring to me, I just redid update again after reinstalling 2.0 as above post and 2.1.1 works, but this time I made sure octoprint was up in the web browse.

thegarbz, 2020-10-25:
@carl1961 I too saw the script error, but after following the instructions (restarting) I haven't seen it since.

UnchartedBull, 2020-10-26:
ok seems like everything is working for everyone now. Closing this here, if you still experience issues please comment / reopen.

@thegarbz good idea, updated the release notes for 2.1.1.

closed 2020-10-26 by UnchartedBull