UnchartedBull/OctoDash #1127: Filament Change on MK3s

jjuuzzii, 2020-10-25:
Filament Change via OctoDash isn’t working anymore without the Prusa knob. (M600 Filament Change is disabled in OctoDash) 

In 2.1 I could unload and Load filament via the application. 
Since 2.1.1 my Prusa always triggers the M600 to load, unload and Purge, I think.

Again M600 in the Application is disabled! 

 - Hardware: Raspberry 3b, hyperpixel 4.0
 - OctoDash Version [v2.1.1]

UnchartedBull, 2020-10-26:
Did you update your Printers Firmware as well? I haven't changed anything regarding Filament Change in release v2.1.1. I think I've read somewhere that the Prusa printers trigger M600 on their own if they detect a filament change.

Protoncek, 2020-10-28:
I played with that, too. At the end i found out that best way to change filament in Prusa is using M600. Extruder heats up and extruder nicely comes to front right,after that i just press knob to unload. You HAVE to be behind printer anyway, so i don't really see the point of not using prusa's knob. I guess that this complex function is more or less intended for other printers without M600 function...

UnchartedBull, 2020-10-29:
I guess it depends on the user. My Ender 3 (with Marlin 2) also supports M600 but I still prefer the OctoDash alternative just because you don't have to swap between 2 systems and I like the design more. 

Did you get anywhere with disabling the Prusa Filament Change Detection though? I do remember that we talked about this but I can't find the OG issue anymore.

closed 2020-11-10 by UnchartedBull