UnchartedBull/OctoDash #1128: FilamentManager spool data in print screen and change filament screen

davide-molino, 2020-10-25:
A cool addition to Octodash can be the percentage and  remaining weight of the filament based onto FilamentManager calculations visible during the print or the filament change menu. You think it's a possible addition?

UnchartedBull, 2020-10-26:
have to check whether there is enough space for this information. Keeping it at the back of my head for now.

Frix-x, 2020-10-26:
As I asked on the issue https://github.com/UnchartedBull/OctoDash/issues/676#issuecomment-714340653, maybe the best compromise would be to add all the currently loaded filament information in the main "Filament" screen of Octodash.
 Then, we could only add main spool parameters on the print screen and add a "more info" popup if you click on it ?

I'm not that good in CSS but maybe I can draw some visual on Illustrator/Photoshop to explain what I mean ? Is this helpfull for you ?

UnchartedBull, 2020-10-26:
Please go ahead and add mockups (create them in whatever you like - as long as png/jpg can be exported), that would help tremendously.

Frix-x, 2020-10-27:
Hi, I'm back with some mockups that I've done very quickly today.
So, two pictures:
  1. The first one is a rework of the main screen during a print. As you will see, I tried to keep the maximum of your work and design principles that I really like !
Left is more centered on the current print state and right is more centered on info like filament and temperatures.
  2. The second picture is a screen that could be used as a "main filament screen". On the bottom right corner, there is a "change" button that would bring you to the current behavior of Octodash to change the filament.
In place of the blue rectangle, we can add all the other usefull information that the plugin FilamentManager or any other else could bring us.

It's a draft, so feel free to modify it as you want :)

[image]
[image]

UnchartedBull, 2021-03-02:
Thanks for all the mockups! Completely forgot to respond to them! They look really good! Need to postpone this to after v3.0 though since I don't have too much spare time right now to work on Octodash.

Frix-x, 2021-03-02:
Thank you, feel free to add them as you want in Octodash :)
I'll be available to discuss and test it when you'll be ready !