UnchartedBull/OctoDash #1131: [BUG] 2.1.1 will not connect to OctoPrint - 2.0.0 on same system connects with no problems.

ManuelMcLure, 2020-10-25:
**Describe the bug**
2.1.1 on AMD64 Debian will not connect to OctoPrint. 2.0.0 on the same system connects fine.

**To Reproduce**
1. Update from 2.0.0 to 2.1.1 using the official .deb and `dpkg --install octodash_2.1.1_amd64.deb`
2. Delete the configuration: `rm -rf .config/octodash`
3. Start up OctoDash
4. Scanning for OctoPrint fails.
5. Accept the default localhost on port 5000 - no connection.
6. Try localhost on port 80 - no connection
7. Try <machine IP> at port 80 - no connection
8. Try "::1" at port 5000 - no connection
9. Try "127.0.0.1" at port 5000 - no connection
10. Reinstall 2.0.0 with dpkg
11. Delete the configuration as in step 2.
12. Restart OctoDash - connection to localhost at port 5000 succeeds.

In all cases the error is
```
Can't connect to OctoPrint
Check the URL/IP and make sure your firewall allows access to port <port> on host <host>
```
where <port> and <host> are what I configures (such as "localhost" and "5000").

I had OctoPrint running in a browser on my tablet in case I was missing an application token prompt but I saw nothing.
As well as the official 2.1.1 .deb I also tried building OctoDash myself with the same results.

**Expected behavior**
2.1.1 connects to the same URL as 2.0.0.

**General Information:**
 - Intel J1900 based mini-PC
 - Debian Buster x64
 - OctoDash Version: 2.1.1
 - OctoPrint Version: 1.4.12

**Additional context**
I originally ran into this trying to run a patched version of 2.1.1 with the DLP dependency removed. When this failed I installed DLP and tested the official .deb with the same results.

ManuelMcLure, 2020-10-26:
I was unable to find any options for a debug log for OctoDash that might provide more info - is there such a thing?

UnchartedBull, 2020-10-26:
This error pops up if the OctoPrint JS Library can't be loaded from the server. Could you make sure that your printer is turned on and connected to the printer and check if that maybe fixes the error?

You can also check if the OctoPrint script is available if you open: <printer:port>/static/webassets/packed_client.js. If you see a lot of code on the screen the script is available.

ManuelMcLure, 2020-10-26:
Thanks  - that got me onto the reason for the issue. I had
```
    devel:
      webassets:
        bundle: false
```
in my OctoPrint `config.yaml` because I was doing plugin development. Once I removed that the connection succeeded.

It might be a good idea to document that OctoDash won't work if that option is enabled.

closed 2020-10-26 by ManuelMcLure

UnchartedBull, 2020-10-26:
Didn't even know that this option existed, thanks for the hint! Added to the wiki: https://github.com/UnchartedBull/OctoDash/wiki/Troubleshooting#cant-get-octoprint-script