UnchartedBull/OctoDash #1134: Octodash segfaults after 'sudo apt full-upgrade'

Aulos, 2020-10-26:
**Describe the bug**
After I upgraded octopi through ssh octodash stopped showing up. When I tried to execute it manually I get a segmentation fault. I was trying to update octodash manually (I had 2.1.0), but after update to 2.1.1 it is still failing.

Upgraded packages:

- libraspberrypi-bin 
- libraspberrypi-dev
- libraspberrypi-doc
- freetype2-doc
- raspberrypi-kernel
- raspberrypi-bootloader
- libfreetype6-dev
- libraspberrypi0
- libfreetype6

**To Reproduce**
Steps to reproduce the behavior:
1. SSH to octopi
2. call `sudo apt update`
3. call `sudo apt full-upgrade`
4. restart rpi

**Expected behavior**
Everything works as before - octodash is 

**Screenshots**
Updated packages - I think it happended after my upgrade on 2020-10-23, but it could have happended after upgrade on 21st
[image]

[image]

**General Information:**
 - Hardware: Rpi 4 2GB with Hyperpixel4 screen
 - OS Info: OctoPi 0.17.0
 - OctoDash Version v2.1.1
 - OctoPrint Version 1.4.2

UnchartedBull, 2020-10-26:
have you started a window manager before executing `octodash`?

Aulos, 2020-10-26:
It doesn't start

UnchartedBull, 2020-10-29:
Could you please double check your `~/.xinitrc`? The segmentation fault while starting octodash from the command line is normal behavior currently.

If you want to start it manually you have to use `startx`. My best guess here is that your full-upgrade somehow wiped the `~/.xinitrc`.

Aulos, 2020-10-29:
It looks pretty normal:
```
#!/bin/sh

xset s off
xset s noblank
xset -dpms

ratpoison&
octodash
```

Aulos, 2020-10-29:
Ok, when I connected keyboard to pi4 and ran `startx` in console `octodash` started - mouse cursor is visible, but other than that it work as expected.

Any idea how I can make sure X server starts after every boot? I've tried to set login to desktop in `raspi-config`, but it doesn't seem to help.

UnchartedBull, 2020-10-29:
Follow these steps and check if everything is right: https://github.com/UnchartedBull/OctoDash/wiki/Installation#start-octodash-automatic-on-boot

UnchartedBull, 2020-10-29:
Oh but since you use fish you may have to check whether it's using a different file.

Aulos, 2020-10-29:
Oh wow, I know what happened - I changed the default shell to fish and its startup script doesn't start X server... After adding it to `fish.config`it started, but mouse cursor is visible. 

Is there any way to turn it off?

Aulos, 2020-10-29:
Yup, I changed default the default shell and X is not starting - when I added it to `fish.config` X starts as it should

Aulos, 2020-10-29:
For interested, this is what I've added to `config.fish`:

```fish
if test -z "$SSH_CLIENT" -o -z "$SSH_TTY"
  xinit -- -nocursor
end
```

closed 2020-10-29 by Aulos