UnchartedBull/OctoDash #1137: Configuring Ocotdash for first time cannot find Octoprint

critter42, 2020-10-26:
**What doesn't work?**
I am trying to go through the first-run install and it cannot find the instance of Octoprint I want, but I can bring it up just fine in my laptop browser. It will find the other Octoprint I have on the network, but even when I shut that one down, it still will not find this one. 

**What did you already try?**
I tried to enter it manually, but all I can do without a keyboard is long press and select cut or copy. I cannot enter any IP info.  I tried looking in the issues and found reference to config.json, but I cannot locate that file on this install

**General Information:**
 - Pi 3B v2
 - Octopi (Rasbian Buster)
 - OctoDash 2.1.1
 - OctoPrint 1.4.0

**Additional context**
I just need to know if I can enter the necessary info via an SSH session rather than the interface some way.

UnchartedBull, 2020-10-26:
https://github.com/UnchartedBull/OctoDash/wiki/Setup-&-Settings#setup-without-keyboard

critter42, 2020-10-26:
Thanks! FYI, Your README.md needs it's "No Problem" link updated then - https://github.com/UnchartedBull/OctoDash/wiki/Installation#setup-without-keyboard  is what it has instead of the link above.

critter42, 2020-10-26:
the xdotool link on that page is broken or the site no longer exists

Protoncek, 2020-10-28:
You can temporary plug into Pi's USB port just about any keyboard you find and it works. There must be a keyboard in your house you can use? It's really the simplest way...

UnchartedBull, 2020-10-29:
Updated both links. Thanks for mentioning!

closed 2020-10-29 by UnchartedBull