UnchartedBull/OctoDash #1140: No api

stevenverde, 2020-10-28:
I just got octodash installed and during the setup screen it tries to connect to octoprint but it says my firewall is blocking it to try port 80 and for the life of me I can't get it to work.
I already tried port forwarding the raspberry Pi .

I'm using a raspberry Pi 3 b+ with a 3.5 SPI touch screen I have access to ssh only using the ip address not octopi.local. and octopi is the newest version also with the newest version of too Buster

eliteSchwein, 2020-10-28:
Use 127.0.0.1:5000

stevenverde, 2020-10-28:
> Use 127.0.0.1:5000

I have tried that and still nothing you mean try using it on the octodash?

eliteSchwein, 2020-10-28:
Yes

stevenverde, 2020-10-28:
Ya I've tried with no luck even tried changing the port to 80 and 5000 still nothing. When octodash loads it shows up on the main screen but when I click it it says firewall is blocking it and if I enter manually even with the IP you gave me still same thing

stevenverde, 2020-10-28:
[image]

stevenverde, 2020-10-28:
[image]

eliteSchwein, 2020-10-29:
Try it with 127.0.0.1:80

UnchartedBull, 2020-10-29:
Also make sure that your printer is connected to OctoPrint and you haven't disabled the packed_client in OctoPrint.

stevenverde, 2020-10-29:
Where would I go to disable the packed_client... Sorry I'm new to all this

stevenverde, 2020-10-29:
Nevermind I got it working thankyou uncharted I'll and eliteschwein for all the help

eliteSchwein, 2020-10-29:
No Problem, can you please close this issue?

closed 2020-10-30 by UnchartedBull