UnchartedBull/OctoDash #1144: Octodash constantly timing out when trying to retrieve files

pilotw7, 2020-10-29:
**What doesn't work?**
When I try to go start a print, Octodash constantly times when I go to select files, majority of the time timing out and saying try again when I try to view the file list. 

**What did you already try?**
I tried restarting my machine and the Pi several times to see if it was an issue with the instance, still errors out constantly. 

**General Information:**
 - Hardware Prusa i3 Mk3s with Raspberry Pi 3b+
 - OS Info Octopi version .017.0
 - OctoDash Version 2.1.1
 - OctoPrint Version 1.4.2

**Additional context**
This happened a few times before on version 2.0.0 of Octodash, but now is happening frequently after I updated to 2.1.1

UnchartedBull, 2020-11-03:
Do you have an SD card in your printer and initialised it in OctoPrint?

pilotw7, 2020-11-03:
Yes, but this happens when I try to view files that are saved to the Pi, when I take out the SD Card the same issue occurs. Octodash is able to load the folder select between choosing SD Card or Pi fine, but when I try to enter selection of files on the Pi then it times out.

UnchartedBull, 2020-11-04:
The problem here is that OctoPrint isn't answering the request from OctoDash within 5 seconds, thus OctoDash thinks a timeout occurs and errors out. I'll have a look at that again, but I don't think there is much I can do.

Did you change anything else while upgrading to OctoDash 2.1.1?

Kinetik1703, 2020-11-04:
Hi,
I just joined Github to give you my comment here as well.

I do have the exact same issue with my fresh setup Octodash...

I cannot load any files from either Card nor internal of the Pi.
It just keeps getting me these time outs.

Sometimes it works and I see the folders internal and card, but when clicking on them, I still have the timeout again.
Can't you change the timeout time here or something like this?

I have no idea about the programming, just guessing ;)

pilotw7, 2020-11-04:
The only thing i can say is that i'm running the non-DLP Fork version from jneilliii, but this problem occurs on my other printer as well that runs the standard version of Octodash, not as frequently however.

UnchartedBull, 2020-11-04:
I'm bumping the timeout to 10s for now. This might fix some of the timeouts, it might not fix any. With the next minor release (v2.2.0) the OctoPrint JS Client will replace the API calls that are currently being used. I'm hoping this fixes the timeout issues.

pilotw7, 2020-11-05:
Will the timeout increase be pushed via an update?

UnchartedBull, 2020-11-09:
Most likely yes. I'll let you know here once it's ready.

thegarbz, 2020-11-10:
Just to add my 2 bobs worth, I frequently get the timeout while another plugin is still working in the backgrounds. Specifically I get timeouts if I just upload a fresh file to Octoprint and in the background the thumbnail extractor and various other plugins which look through the gcode are still operational. Critically this coincides with the inability to properly load the gcode file on the octopi website either so for me at least not an octodash issue.

UnchartedBull, 2020-11-17:
That's quite interesting. Thanks!

UnchartedBull, 2020-11-17:
v2.1.2 has just been released. Please report if that made the issue better or worse :)

closed 2020-12-04 by stale[bot]