UnchartedBull/OctoDash #1146: Touch function not working

DanielP5433, 2020-10-30:
**What doesn't work?**
Touch function on Adafruit 3.5 tft (GPIO version) doesn't work, the firs time set up shows on screen but not responding to touch even though the screen was able to be calibrated.

**What did you already try?**
Followed the install guide with fresh install of Octoprint and drivers/set up from Adafruit.

**General Information:**
Raspberry Pi 3B+ with Adafruit 3.5 inch touch screen (GPIO style not HDMI)
 - OctoPi v0.17.0
 - OctoDash Version v2.1.1
 - OctoPrint Version v1.4.2
 - Python v3.7.3

**Additional context**
I am a complete noob with Python so if I have missed anything please let me know and I will update with more details. I watched the video from Chris Riley and he said that some GPIO screen can have trouble, if so is there anything I can do or should I just but a HDMI screen?

Thank you in advance!

mechan007, 2020-10-30:
I have the same problem but 5 "display

UnchartedBull, 2020-11-02:
I don't have a GPIO connected screen (and don't recommend them either, since most of them have a very slow refresh rate). If @mechan007 solution doesn't work please browse through closed issues and the troubleshooting guide and check whether you can find a solution there.

It might also be a good idea to follow a tutorial with another app and see whether you can get touchscreen working there.

closed 2020-11-02 by UnchartedBull