UnchartedBull/OctoDash #1148: OctoDash support multiple instances of OctoPrint

ToledoRayburn, 2020-10-31:
**Is your feature request related to a problem? I have a raspberry PI with 4 instances of OctoPrint installed controlling 4 printers.   I installed OctoDash to the raspberry PI.   It was aware of what instance I wanted OctoDash to control.

**Describe the solution you'd like**
It would be nice to be able to setup OctoDash to interact with all instances of OctoPrint.   The feature would allow you to know what instance is active by displaying the Title from OctoPrint.   There also would be a means to switch between active octoprint instance.   Preferably this would be done with hot buttons on the main screen.

**Link other projects**
If you want OctoDash to include other OctoPrint plugins or similar please link them here.

**Additional context**
Add any other context or screenshots about the feature request here.

UnchartedBull, 2020-11-02:
Probably something for after v3.

kd5ftn, 2021-01-09:
I would also like to see this feature. I have two printers side by side, with the pi running OctoDash in the middle. A left/right arrow, or button to pop up a list of configured Printers/Octoprint instances would be ideal. There would also need to be a persistent title to display what the currently active printer is.

UnchartedBull, 2021-11-03:
Definitely will be possible, the functionality to apply a new config without fully restarting the Pi is already there, so the only thing that is missing for this would be support for multiple configs (and a UI for that). While its not super easy, it isn't super complicated either.

For a first version the UI can also be included in the Custom Actions, so you just would create a new Custom Action there until a fully fledged UI is available for that.

NovaViper, 2021-11-27:
> Definitely will be possible, the functionality to apply a new config without fully restarting the Pi is already there, so the only thing that is missing for this would be support for multiple [...]

Do you need people to test this? I just finished getting my Octoprint setup with two instances for my printers and would be more than willing to give this a try!

GerogeFu, 2022-01-08:
Any update on this? @UnchartedBull

jneilliii, 2022-01-08:
Create copies of config.json to store as templates and edit for each instance (change URL, API key, name, etc. and add appropriate custom action as described below).

```
cp /home/pi/.config/octodash/config.json /home/pi/.config/octodash/config1.json
cp /home/pi/.config/octodash/config.json /home/pi/.config/octodash/config2.json
```
**OCTO1**: add custom action to config1.json
```
                "octodash": {
                        "customActions": [
                                {
                                        "color": "#a93e26",
                                        "command": "OCTO2",
                                        "confirm": true,
                                        "exit": false,
                                        "icon": "print"
                                }
```
**OCTO2**: add custom action to config2.json

```
                "octodash": {
                        "customActions": [
                                {
                                        "color": "#a93e26",
                                        "command": "OCTO1",
                                        "confirm": true,
                                        "exit": false,
                                        "icon": "print"
                                }
```
Install gcode system commands plugin in each instance and set commands as follows:

**OCTO1**:
```
cp /home/pi/.config/octodash/config.json /home/pi/.config/octodash/config2.json -f && cp /home/pi/.config/octodash/config1.json /home/pi/.config/octodash/config.json -f && sudo service getty@tty1 restart
```

**OCTO2**:
```
cp /home/pi/.config/octodash/config.json /home/pi/.config/octodash/config1.json -f && cp /home/pi/.config/octodash/config2.json /home/pi/.config/octodash/config.json -f && sudo service getty@tty1 restart
```

GerogeFu, 2022-01-08:
Thanks for reply, but how to show ip on the screen?

jneilliii, 2022-01-08:
Maybe set the name in config files to match ip? Not sure of the question as it seems unrelated to the request.

UnchartedBull, 2022-02-22:
I just merged the issues into this one so it's easier to track. The functionality isn't implemented yet.

jneilliii, 2022-03-03:
I've just released version 0.0.7rc5 of OctoDash Companion plugin that can handle swapping instances in a little easier way than the hack approach mentioned above. It adds a new `[!SWITCH_INSTANCE]` token that will allow you to change the url field in config.json and restart the service. Once you select the option it will fill in the command field with a `[!WEB]` action, just swap the end with the instance to use, ie `localhost:5000`, `localhost:5001`, `localhost:5002`, so you then have one button per instance.

DimchikUA, 2022-12-21:
Hello everyone! Please add a display of the current IP address to the OctoDash screen. The printer is on a network that I cannot configure. The IP keeps changing and I have to sweat to connect to the web page.
Octopi.local doesn't work for other reasons. A search under "issue" eventually led me here. Want to ask.
Are there plans to add this improvement in the future, or is it already forgotten?