UnchartedBull/OctoDash #1150: npm start script no longer creates dist folder

thegarbz, 2020-11-01:
While working on the status bar I cloned a completely fresh copy of your repo. After running `npm install` and `npm start big` it resulted in an error that the svg files in the dist folder couldn't be found.

After a bit of digging it seems that a dist folder is no longer automatically created by the start script. I'm not sure when this error arose but the above instructions (listed in your CONTRIBUTING.md file) worked fine a few months ago.

Temporary workaround:
execute `npm run-script build` which does create a dist folder. 
After this the npm start commands work just fine.

UnchartedBull, 2020-11-03:
can't really change that. I improved the electron packaging (so now only dist is included), so the icon can't be accessed from src anymore. You only need to run `npm run build` once though, after that the dist folder should be available and `npm run start:big` should work normally. I'll add that to the CONTRIBUTING file.

closed 2020-11-03 by UnchartedBull

thegarbz, 2020-11-03:
Yes correct the build script only needed to be run once. Okay good to know it wasn't a bug :-)

UnchartedBull, 2020-11-19:
Hi!

Didn't find an e-mail of yours, so hopefully this reaches you :)

Really great stuff you keep contributing to OctoDash, thanks! Just wanting to offer the contributed status to you. This would enable you to directly create branches and assign/close/label issues and more stuff.

No pressure though, I totally understand if you'd like to stay a contributor, no hurt feelings here :).

thegarbz, 2020-11-20:
Hey, I didn't realise my email address was set to private. Fixed that, it's now shown on my profile. 

Sure I'd be happy to have contributor status. But I'd limit my involvement to changing labels on issues. I don't want to touch branching on your project since I'm way too new to Github and I don't want to risk breaking something. :-)

UnchartedBull, 2020-11-22:
You should've received an invitation mail from GitHub 👍 

Feel free to create branches though, as long as you don't merge into master without approval everything should be fine. Plus the awesome thing about GitHub is that everything can be reverted :)