UnchartedBull/OctoDash #1156: arm64 install issue

Will-wastelander, 2020-11-06:
**Describe the bug**
Not sure if this is a bug, feature request, or a support request, but I figured I'd mark it as a bug, since it can't be installed via the current installer script.

**To Reproduce**
Steps to reproduce the behavior:
1. Install OctoPi 0.18.0 RC1 arm64
2. Run OctoDash install script
3. Install script fails, appears to be related to installing octodash_2.1.1_armv7l.deb and not octodash_2.1.1_arm64.deb

**Expected behavior**
As OctoPi becomes arm64 friendly, the installer script should account for this.

**General Information:**
 - Hardware [Raspberry Pi 4 4GB]
 - OS Info [OctoPi 0.18.0 RC1 arm64]
 - OctoDash Version [e.g. v2.1.1]
 - OctoPrint Version [v1.4.2]

**Additional context**
releaseURL=$(curl -s "https://api.github.com/repos/UnchartedBull/OctoDash/releases/latest" | grep "browser_download_url.*armv7l.deb" | cut -d '"' -f 4)

eliteSchwein, 2020-11-06:
https://sites.google.com/view/m1dn1ght-3d/downloads/unofficial-octodash-builds?authuser=0

Check this url out

Will-wastelander, 2020-11-06:
I was able to get this working by adding the below to the install script. I'm not the best at writing scripts, but I can try to piece something together to make it work.

I'm not sure if it's the best way to do it, but it works for me.

```
-releaseURL=$(curl -s "https://api.github.com/repos/UnchartedBull/OctoDash/releases/latest" | grep "browser_download_url.*armv7l.deb" | cut -d '"' -f 4)
+arch=$(uname -m)
+if [[ $arch == x86_64 ]]; then
+    echo "x86_64 Architecture"
+    releaseURL=$(curl -s "https://api.github.com/repos/UnchartedBull/OctoDash/releases/latest" | grep "browser_download_url.*amd64.deb" | cut -d '"' -f 4)
+elif [[ $arch == aarch64 ]]; then
+    echo "ARM64 Architecture"
+    releaseURL=$(curl -s "https://api.github.com/repos/UnchartedBull/OctoDash/releases/latest" | grep "browser_download_url.*arm64.deb" | cut -d '"' -f 4)
+elif  [[ $arch == arm* ]]; then
+    echo "ARM Architecture"
+    releaseURL=$(curl -s "https://api.github.com/repos/UnchartedBull/OctoDash/releases/latest" | grep "browser_download_url.*armv7l.deb" | cut -d '"' -f 4)
+fi
```

closed 2020-11-09 by UnchartedBull