UnchartedBull/OctoDash #1158: CustomActions, run a python script to control heater or light.

strangecalling, 2020-11-06:
**Is your feature request related to a problem? Please describe.**
No a problem, but a feature to add.

**Describe the solution you'd like**
I have written a python script to control a heater for a printer enclosure using a Meross smart switch and would like to be able to start it from Octodash.

Currently I ssh in an run the script from a screen session.

The program reads the enclosure temp and switches off/on the heater and has a timer function. I run a light in a similar way also but have not written a program yet.

**Link other projects**

**Additional context**

xommified, 2020-11-10:
I think the feature request could be renamed to something more generic like "Run shell commands with Custom Actions", but seconding this feature request. Unsure of the security implications though?

UnchartedBull, 2020-11-10:
I don't see a security risk here. Everything will be executed by the user that needs to automatically login anyways. So if someone is able to change the custom actions, they also could just exit out of OctoDash and have access to the console.

davide-molino, 2020-11-12:
I use the action Commands plugins, it can link a shell command to a gcode command, so I've inserted the Gcode in the custom actions and it works.

UnchartedBull, 2020-11-17:
@davide-molino that's great! Could you maybe post a screenshot of an example config here (just a GCode linking to a random script / command) so I can put it in the wiki?

Closing this here then since it is possible already.

closed 2020-11-17 by UnchartedBull

davide-molino, 2020-11-17:
I've installed Action Commands Plugin, created my own commands and inserted the GCODE in the config.json section for assign the commands to the custom actions in Octodash.
I use Klipper as firmware so for me the GCODE command look like this:

"RESPOND TYPE=command MSG=action: ..."

For Marlin the command is something ilke:

"M118 //action: ..."

[image]

[image]

UnchartedBull, 2020-11-17:
great. thanks!

strangecalling, 2020-11-17:
@davide-molino Thank you for showing an excellent solution and @UnchartedBull Thank you for taking the time to look into my query.