UnchartedBull/OctoDash #1162: Horizontal green line in filename?

Protoncek, 2020-11-08:
**Describe the bug**
during printing, if Prusa thumbnail is shown there's horizontal green line accross filename

**To Reproduce**
Steps to reproduce the behavior:
Print anything on Prusa MK3S

**General Information:**
 - Prusa MK3S
 - Latest OctoPrint
 - OctoDash Version 2.1.1

**Additional context**
I can't say if it's just my installation... this green line didn't show at previous versions. I didn't install older OctoDash yet to see, though... Line is always shown - at all prints if thumbnail is shown, but not if only percentage is shown.
[image]
[image]

UnchartedBull, 2020-11-09:
Please update your custom-styles.css with the latest NOX theme.

Protoncek, 2020-11-09:
i replaced whole file with one, published here: 
[https://toscode.gitee.com/Shine6Z/OctoDash/tree/master/themes/NOX](url)
 but there's still a green line, although a bit shorter. Is it possible that theme is not yet updated? Did i use correct link?

Protoncek, 2020-11-09:
Hm..
i removed custom style and i think i know why there's a green line: it's graphical "percentage" , since in original theme this line is just below thumbnail...

UnchartedBull, 2020-11-09:
The link you posted is to a repository from another user (the one who is creating the copycat of OctoDash for Bigtreetech). Please use the Stylesheet from this repository. I've updated the progress bar styles there.

Could you please post a screenshot of the not centered issue?

Protoncek, 2020-11-09:
Yesss! 
Now i'm getting somewhere. Now i have saved info about custom styles link into my "notebook" for Printer stuff where i write all important things down (well, i'm over 50 and memory starts to go down, i guess...)
Sorry for my "stupidity", but my knowledge about programming is pretty much basic - literally - i program AVR's in Bascom and i do some small stuff in Linux, but that's pretty much it...
I tried other themes: Glanceable is very nice, sadly it's not black background...

Oh, about "not centered issue": sorry, i wrote that part somewhat stupid - that was happening with non-correct NOX theme. In original one all is correct, and also in all your themes.

Thanks for your help (again..)!

UnchartedBull, 2020-11-09:
Well everyone has their strengths. You would totally embarrass me in anything even closely remote to compiler level. But that is the awesome thing about GitHub, so we can all share our knowledge :).

I can't take credit for any of the themes as they're all created by the community. Maybe if you ask the @thegarbz really really nice and give him some time he might be able to cook something up with a black background (I hope it's ok to link you here). He's been really active lately and just merged his third theme!

If you encounter any other issues feel free to open a new issue 👍

closed 2020-11-09 by UnchartedBull

thegarbz, 2020-11-10:
@Protoncek have a look at the new "Focus" theme that we pushed an hour ago. It's basically the NOX dark theme with the "Glanceable" type bar. Let me know if you like the look and we'll try and get it working for you. I just remembered it also depends on slight change in the Octodash code. 

Maybe if we ask nicely @UnchartedBull can push Octodash 2.1.2 so we get the updated progress percentages in the new themes working ;-)

Protoncek, 2020-11-10:
@thegarbz: yes, thank you! That is what i've been looking for! 
I did some "research" with glanceable theme myself last evening(combining with NOX), but without any usable luck. I did manage to get black background, but then i lost other things... i'm just not in this, what can i say...
I'll try it out later in the afternoon, when i get home.
Why am i so "pain in the a**" about black? Well, i have original 7 inch Pi LCD and original theme gray background is just waaay too bright - it's as i would set brightness up to very max on TV...

UnchartedBull, 2020-11-10:
@thegarbz I see what you did there haha :D I'm working on it, will be out by tomorrow (hopefully).

Do they have multiple revisions of that screen? I'm also running the 7" and don't have any issue with a bright display.

thegarbz, 2020-11-10:
I'm sure there are many variants of 7" screens out there. Mine is pretty bright but a large part of it comes down to the lighting conditions in the room. My room is pretty dark so I'm also a big fan of black backgrounds, which is why I've been using the NOX theme from day one and why I was so interested in theming in the first place :)

@Protonceklet me know if you find anything wrong with the theme, you'd be the first person to use it. I expect however you may find issues with the percentage indicators until Octodash is updated past v 2.1.1

Protoncek, 2020-11-10:
@thegarbz : new theme is great, just what i wanted, but see pics: maybe i have too big thumbnails? How big do you have? I can make them smaller if this is the problem...

@UnchartedBull : i added two pics, one with default theme and other with new focus theme, see how much too bright is first one...

[image]
[image]
[image]
[image]

thegarbz, 2020-11-10:
Yeah so three things there:
1. That percentage counter that is under the image will move to the middle of the progress bar with the next version of Octodash so you always have a percentage.
2. Pictures are in a square frame in the theme and it works best with square pictures. If you can edit your Pursa slicer settings to put out thumbnails at 256x256 then it works best.
3. In the settings of Octodash don't use the round progress bar, only the straight one. Then in your theme if you want a round progress bar just click on the thumbnail and you get a big round indicator with the percentage inside it rather than the small completely useless (in my opinion) thumbnail.

Protoncek, 2020-11-11:
@thegarbz : Great, thanks for explanation. I'll wait, it's no rush...
I never use circular progress bar anyway, i just set it to take a picture of it to post a (potential) problem. I think thumb is too small and ... well, let's just say it's not as beautiful as straight line... And i like having (pretty big) thumbnail in my screen :-)
And, i'll set my thumbs resolution as you suggested. I think that currently i have something like 160x160.

UnchartedBull, 2020-11-14:
Short update: the next release will be slightly delayed. I found an alternative to loading the OctoPrint script (which should remove startup performance) and also want to include that (plus a code rework for the setup screen). Hopefully it will be ready on Monday, sorry for the delay!

thegarbz, 2020-11-14:
Great thanks. I also just updated the theme. The width of the filament selector was bugged so the filament remaining weight wasn't being shown. 
@protoncek download the v0.2 theme when the new update is pushed.

Protoncek, 2020-11-14:
Will do. Many thanks to both of you!

Protoncek, 2020-11-20:
Today i updated OctoDash and also Focus theme. I just started to print (it's still "heating"). So far all is like it should be! 
Many thanks to both: @UnchartedBull  and @thegarbz  !

Best regards from Slovenia!

thegarbz, 2020-11-20:
Glad to hear it's all going well, 
Groetjes from an Australian who lives in the Netherlands :-)