UnchartedBull/OctoDash #1166: OctoPrint (TouchUI) in Custom Action iframe?

xommified, 2020-11-10:
**Describe the bug**
When using a Custom Action with command `"[!WEB]http://localhost/#touch"` to try and access the OctoPrint UI, it gets stuck at the login screen (pressing login does nothing).

**To Reproduce**
Steps to reproduce the behavior:
1. Configure custom action as:
```{                                                  
        "icon": "window-maximize",                 
        "command": "[!WEB]http://127.0.0.1/#touch",
        "color": "#e84118",                        
        "confirm": false,                          
        "exit": false                              
}                                                  
```
2. Restart OctoDash
3. Navigate to control > press custom action button
4. Enter credentials
5. Press login
6. Nothing happens

**Expected behavior**
OctoPrint processes login and proceeds to UI.

**General Information:**
 - Pi 4B, Hyperpixel 4
 - OctoPi 0.18.0 RC1
 - OctoDash v2.1.1
 - OctoPrint v1.4.2

**Additional context**
After investigating on an iframe on my PC, and trying to bypass the issue by enabling autologin on OctoPrint, it seems the chromium (?) backend is blocking various parts of the iframe from working: 
[image]

It produces error in the browser console such as "Error: Failed to read the 'localStorage' property from 'Window': Access is denied for this document" or "Blocked autofocusing on a form control in a cross-origin subframe" and various 404/403s.

If I try on my PC with Firefox, it works fine. I assume changing the browser OctoDash uses is probably not going to happen, but I just wanted to get some thoughts on workarounds to try, whether this could be a feature request, or if anyone else has gotten it to work.

My use case is to be able to use TouchUI as a fallback for accessing OctoPrint functionality that OctoDash does not support.

UnchartedBull, 2020-11-10:
Please check whether you have embedding enabled in OctoDash: https://community.octoprint.org/t/multiple-octoprint-instances-in-one-webpage-html-using-iframe-doesnt-work/18872/3. If it doesn't work with that enabled there is unfortunately nothing that can be done here. Electron is bound to the Chromium Engine and there is no alternative with the Mozilla engine available AFAIK. I do know that the normal OctoPrint Interface can be in fact opened via an iFrame once the embedding is enabled. Maybe TouchUI is doing some funky stuff that prevents the embedding. Might also be worth to check their issues.

xommified, 2020-11-10:
Yeah, the embedding is enabled in OctoPrint settings, and rebooted for good measure. Been chatting in the OctoPrint Discord and others have reproduced the same issue as well, and linked this issue: https://github.com/OctoPrint/OctoPrint/issues/3482#issuecomment-614005423

Seems like there's not much that can be done unfortunately, like you said. Just wanted to get some thoughts on it.

UnchartedBull, 2020-11-17:
Might be a long shot but one thing I can think off is the following:

- Install https://plugins.octoprint.org/plugins/actioncommands/
- Create a GCode there that calls a script which opens Firefox with the correct URL for TouchUI

Two things I'm not 100% sure about are though:
- How are you going to close Firefox again (don't know if ratpoison does show the close button)
- How ratpoison handles the whole thing

Just something to try maybe if you have an hour or two to spare :)

Closing this here, since there is, unfortunately, nothing I can do about that from OctoDash ...

closed 2020-11-17 by UnchartedBull