UnchartedBull/OctoDash #1171: Extruder cools down, after Loading the Filament with Filamentmanager

mawi1986, 2020-11-11:
Hello everybody

I have the current version of Octodash running, installed on the weekend.
But now I have the problem when I load the filament via the filament manager that the extruder temperature drops to 0 after loading, which is of course very unpleasant.

Greetings Martin

zdm-as, 2020-11-16:
me too. same behavior ... also there is no option to unload filament.

octodash v 2.1.1 
octoprint v 1.5rc

UnchartedBull, 2020-11-17:
That's how it works currently. v2.1.2 will set it back to the last temperature.

@zdm-as have you configured the feed length and feed speed?

zdm-as, 2020-11-17:
@UnchartedBull I used the M600 option in the filament settings. (I use the bear = prusa firmware which understands the M600 filament changing code) ... now I also tried to play with the manual settings.  No matter what I do there is still no way to just unload the filament (without loading a new spool). The filament menu has only one option to "choose new filament" (from the given plugin list) or you can skip it (and may skip the following pages too) but at the end you ending up with your printer who still wants to load new filament. 
This whole filament thing does not really works out for me. I mean, it is way to complicated and entering all my filament in this (outdated) plugin by manually adding hex. codes for the colors and then not even the proper temp. for the selected filament is set automatically. And last but not least it also seems that this Octoprint filament plugin has not been developed for years.

closed 2020-11-17 by UnchartedBull

UnchartedBull, 2020-11-19:
I can't change the M600 behaviour (that will always reload filament, since that is what a filament change is). If you just want to unload filament you can always define a custom action for that. The Filament Change Progress (OctoDash and M600) will always assume you want to reload filament after unloading.

Well you don't need to use the Filament Manager Plugin, if you disable it in the settings you won't be shown the selection screen. It is fairly useful though since you can define a temperature for each roll and don't really worry about that in the slicer anymore, but as stated you can completely disable that and not loose out on any functionality (except for what the plugins provides of course). The plugin isn't outdated, Olli took this over (there were a few releases quite recently). There are plans to support SpoolManager soonish which is his newer take on a Filament Manager (still the same basic concept though).

PR #1188 (2020-11-17): Remember last nozzle temperature during filament change
Fixes #1171.