UnchartedBull/OctoDash #1174: octodash nad TFT 3.5 GPIO

jpokorny12, 2020-11-12:
Hi,
trying to install OCTODASH on raspberry. When I connect to an HDMI monitor, I issue the OCTODASH interface. I would like them on a TFT connected via GPIO. Is there any way to get them there?
Thank you

thegarbz, 2020-11-13:
You need to configure Xorg to only output on the TFT. Maybe the bottom of this forum post can help: https://www.raspberrypi.org/forums/viewtopic.php?t=96155

jpokorny12, 2020-11-13:
Thank you for your help.
It didn't help me at all, or I don't know how to use it.
Can you please give me more detailed instructions?

jpokorny12, 2020-11-13:
I use this driver 
https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)?fbclid=IwAR0C7nUfBRI0UTPGu4QUbuTAZGrzi5yfBnaPNODXNplTzpPZPuMYnhnfp2k
-Install octodash
sudo apt-get install xserver-xorg-video-fbturbo

jpokorny12, 2020-11-13:
I have octodash on screen.
But touch is mirrored :(

thegarbz, 2020-11-13:
Mirrored or upside down?

I have that Waveshare display somewhere, it worked out of the box for me. But the input being upside down is a well known problem. There are a couple of fixes https://www.raspberrypi.org/forums/viewtopic.php?t=175100

What exactly are you trying to achieve?

jpokorny12, 2020-11-14:
When i touch left top corner it recognise touch in right top corner

thegarbz, 2020-11-14:
Try some of the fixes in the link I sent. Or try running the calibration as per Waveshare's wiki.  That's all I can suggest.

jpokorny12, 2020-11-14:
Thanks. It works
I use this
```
git clone https://github.com/swkim01/waveshare-dtoverlays.git
sudo cp waveshare-dtoverlays/waveshare35a.dtbo /boot/overlays/
sudo nano /usr/share/X11/xorg.conf.d/40-libinput.conf
```

```
Section "InputClass"
        Identifier "libinput touchscreen catchall"
        MatchIsTouchscreen "on"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
 Option "TransformationMatrix" "1 0 0 0 -1 1 0 0 1"
```

```
Ctrl + o
Ctrl + x
Sudo reboot

```

https://raspberrypi.stackexchange.com/questions/60872/inverted-gpio-touchscreen-using-99-calibration-conf
https://raspberrypi.stackexchange.com/questions/60872/inverted-gpio-touchscreen-using-99-calibration-conf

closed 2020-11-14 by jpokorny12