UnchartedBull/OctoDash #1176: Add support for MMU2 Filament Change as a plugin on OctoDash

Tor-Einar, 2020-11-13:
This is not a major problem, it will just improve Octodash usage for me.

I'd like for Octodash to ask for which filament to use after I have selected Print.

Print -> FIlament Select -> After warmup the filament rutine is run (at least check if the nozzle is prehated)

https://github.com/derPicknicker1/OctoPrint-Mmu2filamentselect

ammas09, 2021-01-17:
I agree with @Tor-Einar I too would love this feature.  I prefer to start prints from my physical printer (the main point of my using OctoDash).  That being said this is not a deal breaker.

Montezuma-69, 2021-08-21:
+1 here. I always have to wait until the timeout, just after that I can select it on the original LCD.

UnchartedBull, 2021-08-23:
Is that selection coming up as an action command prompt in OctoDash? These command prompts will be also shown in the next version of OctoDash, so that should probably make that a lot easier.

AndersWinther-Dahl, 2022-09-26:
Hi @UnchartedBull
I'm working on making this possible via the PrusaMMU plugin for OctoPrint (https://github.com/jukebox42/Octoprint-PrusaMMU) 
Status right now, is that it works pretty well:
[image]

Before cleaning the code and make it ready for a PR, I would like to know if you're interested in merging it in?

Thank you for a great piece of software!

Best regards,
Anders

AndersWinther-Dahl, 2022-10-01:
If anyone is interested, you can try out my build of OctoDash.

Make sure to install the Prusa MMU plugin in OctoPrint (https://github.com/jukebox42/OctoPrint-PrusaMMU). 
Please note that it's not the same as originally asked for in this issue.

Download my build of OctoDash from here:
https://drive.google.com/file/d/1lQc72k-ee_bilrGrWe074PZGInpsLS1N/view?usp=sharing

Place it on your Raspberry Pi (tested with Raspberry Pi 3 Model B) and install it, it will take some time, so have patience:
`sudo dpkg -i octodash_2.3.4_armv7l.deb`

Restart the Pi:
`sudo shutdown -r now`

If you start from scratch, you can enable the plugin on startup, otherwise you can do it in settings > plugins > Prusa MMU

Here you can see it in action, from clean install:
[image]

Please leave feedback :-)

FieldofClay, 2022-10-31:
I've been using your Octodash version @AndersWinther-Dahl the past few weeks, it works well. Only thing I've noticed is it doesn't pull the spool details from Spool Manager. Each option is blank.

AndersWinther-Dahl, 2022-10-31:
Hi @FieldofClay 
That's correct, at the moment it only uses info from the "Prusa MMU"-plugin itself.

I don't expect it would be hard to get the detailt from Spool Manager. 

I'll look into it.

Absolucy, 2022-11-11:
Here's an updated version of @AndersWinther-Dahl's build (I pulled the latest commits from the main repo), including arm64 and amd64 builds (because imo it's stupid to use 32-bit apps/distro when you're using a modern Raspberry Pi): https://drive.google.com/drive/folders/19QNzVRq1o2RVDOdIMtUnzRXzg9-_RDzH?usp=share_link

AndersWinther-Dahl, 2022-11-12:
Hi @FieldofClay
I did some changes, to get the spool details from SpoolManager / Filament Manager.

You can get my new build from here
https://drive.google.com/file/d/1G3SraqhyZNEmullKsh-m1EqkR1lyr3jx/view?usp=drivesdk

Hope it works for you.

Best regards,
Anders

Absolucy, 2022-11-13:
@AndersWinther-Dahl access denied. also, can you post arm64 builds (rather than just armv7 builds)? I'd build them myself, but I always get an error about it being unable to import 'conf' when I run a build I made locally

AndersWinther-Dahl, 2022-11-13:
@Absolucy 
Yeah, I guess I didn't share it correct. I made builds for both arm64 and armv7 now.

arm64: https://drive.google.com/file/d/10-okil1mDdHBarBCA6eg_-50QYHqFVba/view?usp=sharing
armv7: https://drive.google.com/file/d/1W7-eeDYrxfyRL5c8YjKMxs7QUa2sGeKK/view?usp=sharing

Tor-Einar, 2022-11-13:
I love to see that this is coming alive :)

But I can't get this to work. Well it does work great from my computer as a plugin on octoprint interface, but can't get it to work directly on my raspberry pi, 

When I try to print driectly from Octodash, I never get the dialog box up, (but my octopod app on mu iphone tells me that my printer is waiting on a user input), going to my computer, the pop-up is  there on my screen.

This is my installation, no error messages. any suggestions?
[image]

AndersWinther-Dahl, 2022-11-14:
You need to enable PrusaMMU on the Octodash - screen.
If you're using SpoolManager or Filament Manager, you should also enable one of those.
In the OctoPrint web-interface, you should configure the PrusaMMU plugin, to tell it where it should get filament info from.
Hope that helps

Tor-Einar, 2022-11-14:
> You need to enable PrusaMMU on the Octodash - screen. If you're using SpoolManager or Filament Manager, you should also enable one of those. In the OctoPrint web-interface, you should configure the PrusaMMU plugin, to tell it where it should get filament info from. Hope that helps

Hi,

No it is ok, it was a firmware dialog that caused the issue. 

But thanks for the reply

FieldofClay, 2023-01-25:
@AndersWinther-Dahl's build has been working perfect for me for the last few months with 50+ prints, I have not had any issues with it. I can't imagine Octodash without it.

As far as I can see from a user perspective, this is stable and a worthy candidate to be included.

DittJ, 2023-02-06:
@AndersWinther-Dahl thank you for your post. I am new to all this and wanted to ask if you could tell me what you mean exactly by downloading your octodash and placing it on my raspberry pi. I tried it but it doesn't work. Is there a special place where I have to save the file to execute it ? It would be really nice if you could help me. Thank you in advance

jneilliii, 2023-02-06:
@DittJ for a pi device you would download the armv7 file linked above. Then transfer the downloaded deb file to your device (I personally use FileZilla for this, but you can also use winscp). Once the file is transferred you have to install it, SSH to the pi and then from the folder where you transferred it to (typically /home/pi/) you run the command 

```
dpkg -i /home/pi/octodash_2.3.5_armv7l.deb
sudo service getty@tty1 restart
```

DittJ, 2023-02-06:
Thank you for your help. Now it works, I am very happy. Thank youVon meinem iPhone gesendet

sudo service ***@***.*** restart

nsolling, 2023-03-18:
Hey - This looks exactly what I am waiting for.
Any idea when it will be merged into the main branch? And if not, can I get access to the build? Unfortunately the google drive links have expried.

Mange tak Anders - Fedt lille projekt

AndersWinther-Dahl, 2023-03-25:
Ahh, yes, forgot that I switched cloud storage.

ARM64:
https://1drv.ms/u/s!Agx1vEbRuI2Tgb0N2N6jdvHT5DNslw

ARMv7:
https://1drv.ms/u/s!Agx1vEbRuI2Tgb0LS8kJ4KD0ImXH-A

Sorry for the late reply!

Kevman323, 2023-09-05:
@AndersWinther-Dahl The latest version of the PrusaMMU plugin breaks the filament selection. Is it possible to get an updated version, or the source code?

AndersWinther-Dahl, 2023-09-05:
@Kevman323
My fork can be found here
https://github.com/AndersWinther-Dahl/OctoDash/tree/main
When I find some spare time I'll look into it myself.

sirCarljam, 2023-10-02:
I would really love PrusaMMU plugin in OctoDash.