UnchartedBull/OctoDash #1177: Make the standby screen subscribe to PSU control

mikekscholz, 2020-11-14:
It would be nice if the PSU control integration worked both ways, and would connect back to the main screen when PSU control is triggered remotely from the web interface instead of from the touchscreen. Ive started tinkering and looking for a way to do this but with your deeper understanding of typescript you may find the solution way easier than I can.

UnchartedBull, 2020-11-17:
I suppose this will be done in v2.2.0 once the web sockets are used. I just checked and there is actually a message being sent for connecting and the connected state. This would then easily enable OctoDash to switch between states in both directions without any interaction from the user.

`{"event": {"type": "Connecting", "payload": null}}`
`{"event": {"type": "Connected", "payload": {"port": "VIRTUAL", "baudrate": 115200}}}`

Just leaving these here for later :)

DisasteR, 2021-02-11:
I would love to have my octodash waking up without the need to press on screen :D have you some ETA for this ?

UnchartedBull, 2021-02-12:
Next release, so about ~2 weeks if everything goes to plan

closed 2021-03-18 by UnchartedBull

DisasteR, 2021-03-26:
Hi,

Are you going to release soon ?

Regards

UnchartedBull, 2021-03-26:
Beta is still running, will probably a week or two.

PR #1438 (2021-03-18): Switch to websockets
Not ready to merge yet. Sorry if you got all excited :/. This (currently) is just to keep track of what issues will be closed by this huge PR. If you have any thoughts - feel free to comment

Closes #1358 
Closes #1123 
Closes #595 
Closes #1177 
Closes #1374 
Closes #1505

Fixes #1304 
Fixes #1086 
Fixes #1062
Fixes #1447
Fixes #941
Fixes #1340
Fixes #1550