UnchartedBull/OctoDash #1179: Getting stuck with Xinit giving up and a server error

Doormat1, 2020-11-14:
I have octoprint 1.4.2 installed and have run the latest install script correctly, or at least putty says I have. I go to my pi with an 3.5inch LCD screen correctly installed. (I did have touch UI running on 1.3.12 fine but didnt like the look so I wanted a change)
Now this is what I get on my screen. I have tried the troubleshooter steps and this is what I get in Putty - 
[image]

[image]

UnchartedBull, 2020-11-17:
It seems like something with an UI is starting up before OctoDash (thus the screen is already in use). Maybe something that is leftover from TouchUI? Could you please double check whether everything got removed properly? Please also post the contents of `~/.xinitrc` and `~/.bashrc` (only the line were xinit / startx is being called) here

Doormat1, 2020-11-17:
Im sorry to waste your time but it seems that the mistake I made was to run
the installation script as sudo. Reinstalled without sudo and its working
fine now. I feel like an idiot.

UnchartedBull, 2020-11-17:
No problem :) thanks for the follow-up!

closed 2020-11-17 by UnchartedBull