UnchartedBull/OctoDash #1182: OctoDash Error

fireguy079, 2020-11-16:
i keep getting an error on my OctoDash about "Cant Retrieve layer progress" Http failure response for http://octopi.local:80/plugin/displaylayerprogress/values: 0 Unknown Error.

thanks,
Kevin
[image]

UnchartedBull, 2020-11-17:
error 0 means that OctoDash isn't connected to OctoPrint and can't reach OctoPrint either. please make sure that OctoPrint is up and running and that the URL configured in OctoDash is correct.

Which version of OctoDash are you running?

fireguy079, 2020-11-17:
> error 0 means that OctoDash isn't connected to OctoPrint and can't reach OctoPrint either. please make sure that OctoPrint is up and running and that the URL configured in OctoDash is correct.
> 
> Which version of OctoDash are you running?

Hello,

Version is 2.1.1 and is it better you use static ip and use the in OctoDash URL?

fireguy079, 2020-11-19:
issue fixed when i updated OctoDash to use the IP address of the pi.

closed 2020-11-19 by fireguy079

manipel, 2021-01-31:
Changing the url in config.json did not work (localhost, ip-adress). Removed and reinstalled DisplayLayerProgress pluging did do the trick. When removing from Octoprint also clear data.