UnchartedBull/OctoDash #1184: Server api error 500 , can't retrieve folders / files

zdm-as, 2020-11-16:
I installed octodash and all the corresponding octoprint 3rdparty plugins yesterday. Installation and setup went smooth. The problem is that I can't access a single gcode file - always comes up with "cant't retrieve folders ..." error msg. -> see screenshots. 

All the other menu items just works as expected. I can (or have to) upload and access my gcode files via octoprint web interface (or via the OctoPod app) and when the print is running the current file is correctly shown at the octodash screen (percentage, print time, layer progress, even the Prusa thumbnail preview) - everything except the access/load file menu works fine :)

OctoDash 2.1.1
OctoPrin 1.5rc
RaspberryPi 4B
SanDisk SD card 64GB (only a few folders & files on the medium - not hundreds ...)

[image]
[image]

UnchartedBull, 2020-11-18:
I see that you're running 1.5rc. Which rc exactly are you using? There was an error in rc1 which was fixed in rc2 according to the release notes:

[image]

Please upgrade to 1.5rc2 and check whether the error still occurs.

zdm-as, 2020-11-18:
You're absolutely right. At the time of writing rc1 was the latest release. I've now installed the rc3 (just released today) and also updated OctoDash to the most recent v 2.1.2. So the api access problem is gone but now only the files in the root folder are listed  - all sub-folders shows empty (0 files).  Probably another rc-related problem (yet still a remarkable improvement ;) ...

UnchartedBull, 2020-11-22:
I'll check what might have changed with the API or whether that is a bug in the RC :)

zdm-as, 2020-11-30:
@UnchartedBull just fyi I already updated to the 1.5 final release but the sub-folder problem still persists.

seipekm, 2020-12-01:
@UnchartedBull i can confirm

Kawuezel, 2020-12-01:
@UnchartedBull Yes, same here. No files in folders on OctoPrint 1.5

UnchartedBull, 2020-12-01:
I'll have a look at this and publish a patch release today or tomorrow if it's an issue with OctoDash :)

UnchartedBull, 2020-12-01:
Quick update: This seems to be an issue with OctoPrint, rather than OctoDash (at least from what I can tell right now). I opened an issue over in their repository: https://github.com/OctoPrint/OctoPrint/issues/3844. I'll keep this issue open and pin it until this get's resolved (either via OctoDash or OctoPrint, whichever turns out to be the culprit of this)

UnchartedBull, 2020-12-01:
Confirmed bug in OctoPrint. Please have a look at the linked issue for me details (and a possible release date, which could be a while according to foosel)

Kawuezel, 2020-12-01:
Thanks for your help and update @UnchartedBull. So we will have to wait or not use subfolders until this has been fixed.

seipekm, 2020-12-04:
I test now thenew octoprint Version 1.5.1. Subitems are here but the folder counter is 0

UnchartedBull, 2020-12-04:
Ok that is great news! The folder count should be fixed (or removed depending whether it's still feasible) in the next release :)

PR #1259 (2020-12-22): replace folder files info with size info
Closes #1184
closed 2020-12-22 by UnchartedBull