UnchartedBull/OctoDash #1192: Octodash settings only shows the native six custom actions

davide-molino, 2020-11-18:
It's a really cool addition the scrollable custom action section, but in the octodash settings there are only the original six actions. If I add a new action I can't edit things like confirm or exit after execution from the settings menu. It's possible to add all the new actions in that menu ?

UnchartedBull, 2020-11-18:
That's by design currently as described here https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions#configuring-custom-actions (this was squeezed in last minute, so I hadn't had any time to update the settings). This isn't super high priority, so it might take a while to update the settings view.

davide-molino, 2020-11-18:
ok, thank you... The scrollable actions are sick... very useful