UnchartedBull/OctoDash #1193: Filament menu (middle icon) question: how to disable it / disable automatic behavior

MartinMajewski, 2020-11-18:
Hi all,

I would love OctoDash as my companion screen, but the filament menu (middle icon) seems to behave oddly and I cannot understand the benefit of it. It makes my life unnecessarily hard for me while operating my machines.

Maybe you can help to clarify things, please?!

My setup: multiple MK3S (heavily modified) with the original LCD Display, OctoPrint on an RPi (3B+ and 4B), and OctoDash 2.1.2 ontop. Not all displays attached to the RPis are the same size, btw.

I like to use the original LCD screen for machine setup actions like filament changes, preheating, and such. OctoDash on the other hand shall serve as a file explorer and axis jogging controller. The reason is, that I want to rely on the manufacturer's FW features which are specific to the machine and thus reliably available on the machine's interface while print job-related actions are more on the OctoPrint side and thus better suited for OctoDash. That's my point of view, at least.

Now comes the filament menu (not the filament manager, which is not enabled in the plugin section), which frequently annoys me as soon as for various reasons I accidentally tap on it: it starts preheating the machine (to unwanted temperatures by default) and unloading the filament automatically. One is only left with the option to wait for five seconds or to perform this unwanted action immediately. Backing off of this menu does not abort this process. I don't like this behavior at all and I think it is bad UX-design to start such an action without the user's clear permission. An unwanted tap on the middle and therefore most prominent icon can happen any time and it seems like a death sentence with no option to sign out from it.

I don't understand the design decision behind this, nor why it is the most prominent option in OctoDash. It completely breaks the workflow and usability for me, especially on machines with small displays or when connected over VNC. After a tap on the filament menu I am forced to wait at or visit this machine in order to reload the filament, which is stupid af, don't you think?

Is there a way to completely remove this option from OctoDash without having to mess with the source code? Or is there at least a way to disable the automated behavior behind it? It would make OctoDash so much more usable for me. Is nobody else experiencing this inconvenience?

I don't want to attack anyone, just asking as a confused user.

Thank you very much for working on this project!

Cheers,
Martin

thegarbz, 2020-11-18:
It looks like you've configured yourself into a corner. 

When filament manager is disabled tapping the filament menu starts the process for manually changing filament which includes a 5 second timeout and which would allow you to simply hit the back button in the top left without changing anything on the printer. 

When filament manager is disabled AND the option use M600 to change filament is enabled then simply tapping the filament button actually sends M600 straight away to the printer. The printer then waits for the temperature setpoint which comes 5 seconds later and starts it's own filament change process. 

Your best bet sounds like to disable the M600 to change filament. That would give you 5 seconds to abort if you accidentally tap the filament change button. An even better option would be to simply install filament manager. No need to use it but to have the plugin enabled would simply open a menu when you tap filament by accident which you can back out of at any time without negative impacts. 

As for the usability / workflow, I personally consider the filament manager integration a killer feature, it's what drove me to Octodash in the first place and since filament manager needs to know what filament I load in my workflow literally is Turn the power on, > Tap the filament button and select the filament I am going to load. It's a front and centre feature of the UI. 

I'll see if there's anything I can do with CSS to get rid of the button you don't use.

thegarbz, 2020-11-18:
Okay I was bored
[image]

copy and paste the following into ~/.config/octodash/custom-styles.css
```
.main-menu__options td:nth-of-type(2) {
    visibility: hidden!important;
    width:1vw!important;
}

.main-menu__options td {
    width: 42vw!important;
}
```

MartinMajewski, 2020-11-19:
Thank you for your time and reply! I really appreciate it!

I've hidden the icon via the CSS entry now, that is a good workaround for me, thanks!

Regarding the other points you've made:

Sending such a critical command right away to the printer should be optional to be disabled in the config. Furthermore, it is misleading for new users to OctoDash (like myself) when disabling the Filament Manager enables another filament managing behavior in a hidden side-effect manner. Maybe this could be at least mentioned somewhere during the initial setup?! I mean, OctoDash should make life easier not forcing a user to learn all in and outs of the software by heart at day one, in order to not "configuring yourself into a corner".

Maybe it could be made possible in the future to have something like a submenu (the famous "three bars menu") where unwanted options could be moved out of the front view?!

It is, as they say: one man's trash is the other man's treasure. I am printing 95% of my time with PETG and therefore only change spools if they are empty or I need another color that is not present on one of the other machines. Therefore, I don't see the filament manager as a "killer feature" but more of a "feature that kills" - my user experience.

Anyhow, the workaround is great. I want to thank you once more for your efforts and I wish you a good day.

By,
Martin

thegarbz, 2020-11-19:
Yeah I agree but my point is it's a configuration quirk. If you uncheck the M600 option it doesn't send any command right away. After a 5 second delay it changes the temperature which isn't critical and can be reverted and it's not possible to do this mid print.  

If filament manager is disabled there is nothing really "managing" filament, the function is just a process that allows you to unload the old filament. This could maybe be better documented. 

@UnchartedBull there's potential room for improvement here. 
1. Is it possible to change the M600 behaviour so it only sends M600 *after* the 5 second delay to select the temperature? I've never noticed this but @MartinMajewski pointed me to it: If M600 is sent to the printer the printer gives no indication that anything has happened and then suddenly without warning it triggers a filament change.  Currently if you use M600 and click on a filament, and then click back within the 5 seconds you think everything is okay and suddenly 30 seconds later the printer unloads. A better approach would be:
  1.1 User selects filament.
  1.2 Filament page shows the temperature and the 5 second timer.
  1.3 After the 5 second timer expires the temperature is sent to the printer.
  1.4 AFTER the temperature is sent M600 is sent and the printer is told to unload the filament at the correct operating temperature. 

This change would allow the user to abort the M600 command in the 5 second timeout that the filament screen already gives the user, and the temperature change on the printer also makes it clear that the printer is doing something.

UnchartedBull, 2020-11-19:
Maybe a few takes from me on this:

I'm mostly with @thegarbz here. The whole filament thing is one of the most crucial parts of the application for me. Having the temperature being set automatically based on the loaded filament is really useful and dramatically simplifies slicing a model (basically just define whether it's PLA / PETG / TPU / whatever and of you go). 

The unwanted temperature can be configured in the settings (although only one temperature possible atm).

There is unfortunately no way to abort an M600 command from OctoDash. Once the command has been submitted you have to cancel it from the printers LCD. The main problem here is that the M600 command shouldn't be sent directly after clicking the button, but rather after the temperature has been reached, so this is most likely a bug which will be fixed in the next version. This would then leave ample time to cancel the change progress. @thegarbz would there be a benefit to send the M600 command directly after setting the temperature or is it ok to wait until the temperature has been reached?

Regarding the user's permission: The whole filament change menu tries to simplify the process, so asking for permission every step of the way would kind of make it useless. And it doesn't get triggered automatically, but only after you hit the filament icon on the homescreen. I'm with you that the user should be able to cancel it within a given amount of time (at least the 5s). 

Disabling the filament manager doesn't enable anything else. It just skips the first screen where you can select which filament you're going to load.

So this menu isn't stupid for some users. It might be for others, but no one forces you to use it. With the theme from @thegarbz (Thanks!) you can easily hide it. It will still in that place for the foreseeable future though. There probably also won't be a menu to disable certain options from the main screen. This means a lot of effort and I don't have that much time to spare right now, thus I'd like to focus my efforts on other ideas. Feel free to create a PR for that though 👍 

I'm going the change the title of this issue to reflect the M600 Bug.

thegarbz, 2020-11-20:
@UnchartedBull I have some more bug information for you. I was wrong, the problem is not when the M600 gets sent but rather that the filament change process from your UI doesn't abort. M600 already gets sent after the temperature is reached, but there's a but hitting the back button when Filament Manager is disabled:
Here's some examples I tried and the outcome:

* Filament Manager Enabled:
  * Back button tapped during 5 second timeout: - No change to the printer. Correct behaviour. 
  * Back button tapped after temperature command sent: - The printer keeps heating to filament temperature. No M600 command is sent. Not ideal since the temperature change doesn't get cancelled but otherwise okay. 

* Filament Manager Disabled:
  * Back button tapped during 5 second timeout: - After about 2 seconds back at the main menu the printer heats to the filament temperature anyway!! M600 is sent when temperature is reached. This is undesired behaviour. 
  * Back button tapped after temperature command sent: - The printer keeps heating to filament temperature. M600 is sent when temperature is reached. This is undesired behaviour. 

I don't know why, but the "back" button behaves differently when Filament Manager is disabled and causes the printer to heat up anyway, and if M600 is enabled after heating it sends M600. 

So the bug is on the first filament screen and not in the M600 implementation.

MartinMajewski, 2020-11-20:
That's what I wanted to explain initially. @thegarbz did it better! Thanks for investigating further!

UnchartedBull, 2020-11-22:
Seems like i forgot to cancel a timeout somewhere. Next version should cancel heating if back button is pressed and only send M600 command if user is still on the filament selection screen.

thegarbz, 2020-11-23:
Indeed if you cancel the timeout that should be sufficient. The M600 already only sends if the temperature is reached.

PR #1277 (2021-01-04): Rework filament change
Mostly behind the scenes code changes #1187 and some design tweaks. Should make #676 fairly easy now since filament manager is now behind a new interface.

Fixes #1193.
Closes #1219.
closed 2021-01-04 by UnchartedBull