UnchartedBull/OctoDash #1205: Temperature to zero button

LizRidley, 2020-11-21:
I'd been using another TFT plugin but recently moved over to this one and its far superior.

On the temperature set screens would it be possible to have a zero button next to the existing set button please?

It'd just be a bit slicker for example having set the nozzle to heat from the main screen for cleaning its then a bit of a faff to go into the controls to select a cool down button or use the decrease buttons to bring it back down.

Hopefully I've not missed something?
Thanks

UnchartedBull, 2020-11-21:
Just press on the temperature. That will set the temperature to 0 (if one is currently set) and to the configured temperature if currently 0. (Hopefully the explanation makes sense, if don't just click the temperature a couple of times and you'll get the hang ;) )

closed 2020-11-21 by UnchartedBull

LizRidley, 2020-11-21:
Brilliant thank you, thought it seemed a bit odd not being able to do it easily, sorry for being dense.

UnchartedBull, 2020-11-22:
No need to be sorry :) This isn't explained super well so no problem.