UnchartedBull/OctoDash #1206: getaddrinfo EAI_AGAIN

TheLion, 2020-11-21:
**What doesn't work?**
After update from 2.0.0 to 2.1.2, OctoDash can't connect to Octoprint running on other RPi. It displays the error:
```
System Error - please restart
getaddrinfo EAI_AGAIN [HOSTNAME-OF-OCTOPRINT-RPI]
```

**What did you already try?**
Tried IP-address of Octoprint RPi in config file instead of hostname, error changes to:
```
System Error - please restart
connect ENETUNREACH [IP-ADDRESS-OF-OCTOPRINT-RPI]:5000 - Local
(0.0.0.0)
```
Re-installed OctoDash over current install, no change.

**General Information:**
 - Hardware Raspberry Pi 4B
 - OS Info Raspbian Buster
 - OctoDash Version 2.1.2
 - OctoPrint Version 1.4.2

**Additional context**

When I delete CONFIG.JSON and setup from scratch, setup is successful and it works until reboot. No difference if I use hostname or IP-address.
OctoDash RPi is reachable with SSH on IP and can ping both hostname and IP-address of Octoprint RPi from OctoDash RPi

I have 2 Octoprint RPi's, (same hardware, same Ocroprint version, different printers) both display same errors.

UnchartedBull, 2020-11-22:
The `getaddrinfo EAI_AGAIN` error seems to be related to a DNS lookup error: https://stackoverflow.com/questions/40182121/whats-the-source-of-error-getaddrinfo-eai-again. Are you using a local DNS server or VLANs by any chance?

And according to this stackoverflow thread: https://stackoverflow.com/questions/49015634/error-connect-enetunreach `ENETUNREACH` seems to be related to a network issue (most likely network down). This error occurs if the Pi on which OctoPrint is running isn't able to ping the specified host:port of the OctoPrint instance within 60 seconds. 

So please make sure:

* That your OctoDash Pi does connect to a network within 60 seconds and can ping the OctoPrint servos
* That the OctoPrint server on the remote Pi is running or starting simultaneously with OctoDash

TheLion, 2020-11-22:
> So please make sure:
> 
>     * That your OctoDash Pi does connect to a network within 60 seconds and can ping the OctoPrint servos
> 
>     * That the OctoPrint server on the remote Pi is running or starting simultaneously with OctoDash

I do run a local DNS server. And as I can ping the remote (on same LAN) OctoPrint RPi's by name and IP-address, it doesn't seem to be a DNS issue.

Before I updated OctoDash, I started the OctoDash RPi and it worked great. With both OctoPrint RPi's. After the updat, it doesn't. Maybe the 

I tested my DNS with both dig and nslookup command from the OctoDash RPi and it reports the correct IP-address.

Maybe the apt update and apt upgrade messed something up?

UnchartedBull, 2020-11-25:
Version 2.1.0 actually introduced the port check, so probably not related to something with apt upgrade. Did you test it with booting OctoPrint first and then starting OctoDash afterwards?

TheLion, 2020-11-26:
OctoDash and OctoPrint run on separate RPi's. Octoprint is always running. Already tried restarting OctoPrint, but made no difference.
OctoPrint is running fine and reachable and controllable through the web-interface, like normal.

It is so strange, that when I reconfigure OctoDash by removing the config.json file, it works until the next reboot.

Does OctoDash have any logfiles I can check?

UnchartedBull, 2020-11-30:
Hmm that is really really weird. I'm probably going the remove the port checking (since it isn't really needed anymore and might just cause trouble like this). Please stay on v2.0.0 until the next release. Sorry!

TheLion, 2020-11-30:
> 
> 
> Hmm that is really really weird. I'm probably going the remove the port checking (since it isn't really needed anymore and might just cause trouble like this). Please stay on v2.0.0 until the next release. Sorry!
If there is anything I can test, just let me know.

Is there a way to downgrade to v2.0.0 ?

UnchartedBull, 2020-11-30:
Yeah I'll have a look at that and let you know, thanks!

Just download the v2.0.0 from the releases and `sudo dpkg -i <file>`. You might need to adjust your config though (OctoDash will tell you if something and what's wrong)

closed 2020-12-17 by stale[bot]

TheLion, 2021-04-04:
Any update on this? 

Just tried 2.1.2 again, but it still shows the same error.

UnchartedBull, 2021-04-06:
Please try the beta linked in the discord, or wait until v3.0.0 is officially released.

TheLion, 2021-04-16:
> 
> 
> Please try the beta linked in the discord, or wait until v3.0.0 is officially released.

The beta works flawlessly.
Thanks!