UnchartedBull/OctoDash #1208: How to change the theme?

titis14, 2020-11-22:
**What doesn't work?**
Hello,
How to change the theme? I would like to put Nox but I do not see or change. I currently have the default theme.
Thank you

**General Information:**
 - Hardware Raspberry Pi
 - OS Info Raspbian Buster 
 - OctoDash Version [e.g. v2.1.2]
 - OctoPrint Version [e.g. v1.4.2]

UnchartedBull, 2020-11-22:
Please read the wiki: https://github.com/UnchartedBull/OctoDash/wiki/Custom-Styles

closed 2020-11-22 by UnchartedBull

titis14, 2020-11-23:
ello, I don't want to customize the theme, I would like to activate the Nox theme because I do not have the same display style and in the files I see that there are 2 themes, how to choose one or the 'other? Thank you

UnchartedBull, 2020-11-25:
You'll need to copy the contents of the provided css files to the `custom-styles.css` as stated in the link above (https://github.com/UnchartedBull/OctoDash/blob/master/themes/NOX/custom-styles.css for NOX).