UnchartedBull/OctoDash #1210: Feature request: Custom buttons on the print control screen

HerrHofrat, 2020-11-23:
I recently switched from Touch UI to Octodash and I really love it! Thanks for this great piece of software! 
I've a led strip on my printer that is controlled with a M355 gcode command - for that it would be really great to have a couple of customizable buttons on the print control screen, alongside cancel/pause/adjust. Maybe 1-3 buttons would be nice, customizable like the ones in the control panel. 
Let me know what you think!

UnchartedBull, 2020-11-25:
I'll have a look at that and check whether there is some nice way to integrate that. Most likely it'll be a link to the already existing Control Screen.

UnchartedBull, 2020-11-25:
Duplicate of #828.

closed 2020-11-25 by UnchartedBull