UnchartedBull/OctoDash #1211: Octodash stuck on installing update screen

hmsq82, 2020-11-25:
**Describe the bug**
A clear and concise description of what the bug is.

Clicked on cog wheel, about update to 2.12 version, this update downloads, once it starts installing it gets 

**To Reproduce**
Steps to reproduce the behavior:
Got to Settings
Click on about
Click update, and wait for process to start
Screen will be stuck for hours on "installing update"

**Expected behavior**
I would expect it to update after clicking update.

**Screenshots**
<a href="//imgur.com/a/YTkd15g">OctoDash stuck on installing update</a>

**General Information:**
 - Hardware [Raspberry Pi 3 Model B Rev 1.2, BTT PITFT50]
 - OS Info [Octopi]
 - OctoDash Version [V0.17.0]
 - OctoPrint Version [V 1.4.2]

**Additional context**
Add any other context about the problem here.

hmsq82, 2020-11-25:
I've now left it overnight night, so 12 hours and it still has not updated and is still stuck on the screen as on the picture I provided above.

Octoprint seems to work fine when I go on the browser, I wonder if this is some sort of permissions, I've followed the manual guide as the script did not work for me.

Is there anyways I can check installation log? Thanks

UnchartedBull, 2020-11-25:
Have you installed the update script? https://github.com/UnchartedBull/OctoDash/wiki/Update#in-app-update (needs to be done manually if you have used on of the older installation scripts)

hmsq82, 2020-11-25:
I just went through the menu on the display, I'll check out that script. It
would be nice if it worked via the screen so no messing about with pc.

Thanks

UnchartedBull, 2020-11-25:
You only need to setup the script once. After that it will work from the screen :)

hmsq82, 2020-11-25:
I just did manual update via ssh, so this is not a bug, and it is a feature. ;)   Thanks for the support!

closed 2020-11-25 by hmsq82