UnchartedBull/OctoDash #1213: Babystepping not working with Klipper?

Whitehawk2000, 2020-11-25:
Hi!

just tried to use Babystepping but the only thing that is changing are the numbers on the display, the Z-Axis does not move.
When using Babystepping with the integrated display with the knob it is working without a problem.
Is it possible, that the Babystepping-function does not work when using Klipper as the firmware?

UnchartedBull, 2020-11-30:
You might need to change the babystepping command that is being used, the standard doesn't work with Klipper AFAIK: https://github.com/UnchartedBull/OctoDash/wiki/Setup-&-Settings#manually-adjusting-the-settings. `zBabystepGCode ` is the one you're looking for. The actual number will be appended directly after the command.

Whitehawk2000, 2020-11-30:
Great to hear, but sadly Klipper obviously won't be compatible to Babystepping with Octodash.
Klipper need the command: "SET_GCODE_OFFSET Z=         MOVE=1".
 Without the "MOVE=1"-command at the end, the Z-height would not be adjusted immediately, but instead at the next layer-change,

UnchartedBull, 2020-12-01:
Mhhh ok and `SET_GCODE_OFFSET MOVE=1 Z=` doesn't work I presume? I'll have a look at that and probably change the command to be something like `SET_GCODE_OFFSET Z=<zDistance> MOVE=1`, where <dist> would get replaced by OctoDash this would work for both cases then I guess, since Marlin would be `M290 Z<zDistance>`. Until then you can maybe create a macro in OctoDash to map M290 to the Klipper alternative.

Whitehawk2000, 2020-12-02:
Hmm, now thats strange... I tried your suggestion and `"SET_GCODE_OFFSET MOVE=1 Z="` only works from the Terminal in Octoprint. A change in the config.json has no effect.
Your hint with the Macro in Octodash made me finally find the recent issues with Babystepping in Klipper. Later i will look at how to write a Macro and if i can sove this problem now by myself. :-)

UnchartedBull, 2020-12-04:
Could you maybe have a look at the OctoPrint Terminal while adjusting the Z-Height and check what commands are received from OctoDash? If it works in the terminal it should work form OctoDash as well. Did you reboot / restart OctoDash after adjusting the config?

Whitehawk2000, 2020-12-04:
Yes, rebooted the whole Pi after every config change. If i am remembering it right, i didn't see anything in the terminal while adjusting. But i can be wrong because the model was relatively complex and the terminal was scrolling through the gcode very fast. Now my printer is occupied. So tomorrow, i will be slicing a simple model and checking it again.

Whitehawk2000, 2020-12-05:
So, now i figured out what happens. 
Everytime i push the button octodash sends the following command:
`SET_GCODE_OFFSET MOVE=1 Z=0.02`
`SET_GCODE_OFFSET MOVE=1 Z=0.02`
.
.
.
But Klipper needs absolut values and not relative ones. So it  is staying at 0.02mm all the time. I think Marlin and other firmwares also need absolut values, so i wonder if or why its working with them.

whoim2, 2020-12-08:
SET_GCODE_OFFSET usually used for ajust after nozzle change in slicer start gcode. For babystep need to use SET_GCODE_OFFSET_AJUST command. But this not need zDistance value, only send command +0.05 or -0.05 (example) for each button, up and down.

Also, maybe use encoder?)
 http://www.ozeki.hu/p_3054-how-to-setup-a-rotary-encoder-on-raspberry-pi.html

UnchartedBull, 2020-12-08:
Thanks @whoim2. According to their docs the mm isn't needed for either command: https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#extended-g-code-commands. Could you try setting the config parameter in OctoDash `SET_GCODE_OFFSET MOVE=1 Z_ADJUST=` (OctoDash will send relative values here, not absolute). Maybe that'll work :)

Whitehawk2000, 2020-12-08:
Oh, that sounds promising. :-) Going to test is it in a few minutes. I wonder why i missed this command , when studying the klipper pages.

> Also, maybe use encoder?)
http://www.ozeki.hu/p_3054-how-to-setup-a-rotary-encoder-on-raspberry-pi.html

Sounds like a nice idea. Maybe it would be better to make an independent plugin for Octoprint instead of implementing it into Octodash???

Whitehawk2000, 2020-12-08:
Oh, yes! It now works like a charm and as its considered to work.
So for everyone else having this problem in Klipper here is a quick summary:

> You might need to change the babystepping command that is being used, the standard doesn't work with Klipper AFAIK: [...]

And there you have to change the code to following:
`SET_GCODE_OFFSET MOVE=1 Z_ADJUST=`

Thank you very much for your help!

closed 2020-12-14 by UnchartedBull