UnchartedBull/OctoDash #1214: choosing preheat temperatures from Octoprint temperature tab

Whitehawk2000, 2020-11-25:
Hi! 
i have stored preheat temperatures in the temperature tab from Octoprint.
But in Octodash there are only the possibilities to choose between one preconfigured preheat temperature or the one if a gcode i loaded. 
In my workflow i often start preheating the printer before i am slicing the stl.
So it would be great if there would be the possibility to choose the temperatures already stored in the Octoprint temperature tab.

thegarbz, 2020-11-28:
So as a workaround / workflow optimisation in the meantime can you pre-heat to say 10deg above the coolest filament you normally print with? Then while you're slicing and playing with your STL you'll have the overwhelming majority of the heating done and you'll be ready to rock and roll quite quickly when you open the gcode file in octodash.

Whitehawk2000, 2020-11-28:
Sorry, but somehow that makes no sense and there is no need for a temporary workaround. I already know which Filament i am going to print. So choosing different temperatures from a list that has already been edited in octoprint would just be more comfortable than tapping the temperature up and down...

Another thing that speaks agaiinst such a workaround is in my case the slow heating of my 12V Heatbed and the fact that i am often printing ABS. That means i need a bed temperature of 110°C. My bed needs around 8min to get from 80°C to 110°C. So with your workflow i would always have to heat up to 110°C no matter if i am going to print materials that need less temperatures. This would be a waste of electricity.

Don't misunderstand me. This is just a suggestion for making the program a lot more comfortable to use.

thegarbz, 2020-11-29:
It was only a suggestion to save you messing around and reduce your heat time. The waste of electricity is practically irrelevant. It takes about 10-20Wh to make up the difference between 80-110C. Back of the envelope calculation says if you're paying German expensive electricity prices here, every 200 prints you can afford yourself a Marsbar. I didn't mean always go to 110 either, I actually meant go to 90C. But whatever.

Whitehawk2000, 2020-11-29:
Hey, i don' wan't to offend you! Just a suggestion to make Octodash more comfortable and usable. There are far better options, than preheating to a random temperature, for example: different gcodes only with the material temperatures for preheating or like i do at the moment, sitting at my PC and choosing the temperatures in Octoprint from my preedited dropdown menu.

Double your calculations on the electricity and then you are right. But with every printer (and other electronics as well) it sums up per year and is definitely noticable on the electricity bill. ;-)

As i said earlier, i don't want to offend you. I really like OctoDash and i appreciate your work. As i am not able to help you with programming i can only make suggestions to make OctoDash better and comfortable. At the moment it is prettier, but lacks some features that were already implemented in Octoscreen.

By the way, another necessary option is to home the X/Y-axis and Z-Axis independently!

Greetings!
Martin

thegarbz, 2020-11-29:
I'm not offended, it's not my project, hence why I was only offering a workaround ;-) The only thing I can take credit for is a couple of the themes.

Whitehawk2000, 2020-11-30:
Ok, then i misinterprated your "But whatever". :-) I like the themes and style. So great work! :-)

UnchartedBull, 2020-12-08:
Hmm this is a hard one, do you possibly have something in mind on how this could look like? I don't really see how this can fit in OctoDash plus I never used the predefined temperatures in OctoPrint. If you can roughly describe how you think this should look like and how many temperatures should fit on the screen (is there a min / max) I can have a look. 
I can't guarantee that I'll work on this though.

Whitehawk2000, 2020-12-08:
The simplest way to realise that would be to add a simple Temperature/Preheat-button, wich opens a configurable submenu with the different temperatures for the different materials. There is no necessity to bind it to the octoprint temperatures, though it would be nice.
Another idea would be to combine it with the filament manager, so it becomes an option to choose for every filament. Other options in the filament manager is something that nobody needs (although it's a nice idea). By the way, i do not know anybody who has got the patience to use such a manager. ;-)

How many temperatures to choose... Thats a difficult question as there are so many different materials to print. At best, so many as the user wants to configure. When the submenu would be something like the listing in the filemanger, there would be no need for limitations.

> I can't guarantee that I'll work on this though.

It is also not an necessary or urgent function. It would just be a nice to have and make the use more comfortable.

UnchartedBull, 2020-12-14:
You need this for the bed an nozzle right? I'm thinking of something like long-pressing the nozzle / bed icon will bring up the temperatures. This way it won't clutter the UI.

Whitehawk2000, 2020-12-15:
That sounds like a good comfortable solution! 
Yes, preheating should be for the bed and hotend. At best with a M190 command, so that the hotend only starts heating when the bed is already at temperature.
I also thought about your question on how many temperatures should be choosable.
I think at least there have to be 3 Temperature ranges. But better would be 5. With 5 ranges everyone, with many different filaments, should be satisfied and able to comfortable configure it for his needs.

UnchartedBull, 2020-12-19:
I think 5 might fit onto the display without a scrolling list. So 5 temperatures each for bed and nozzle it probably is. I'll have to check whether it's possible to retrieve these temperature settings from OctoPrint somehow. If not they need to be defined in OctoDash separetly.

spartanJon089, 2021-02-01:
Not sure if im late to the rodeo or on the right trail, but I think either 
A. add preheat option to the preheat button in the controls menu, kinda like the ender series printers. This will give an option for a few self config preheats for something like pla, petg abs etc. this could possibly pull from the octoprint themp settings as suggested.