UnchartedBull/OctoDash #1216: Raspberry Pi 4 with GPIO 3.5" screen Octodash runs manually from desktop but not console.

nicwilson58, 2020-11-27:
I installed a Pi 4 with the image from your installation site and followed the steps,  I then also followed the troubleshooting, I also looked through similar issues but have not found a solution.

Steps:
Pi4 boots up showing text boot process
Desktop then eventually displays but Octodash does not run.
If you try to run from console it gives a segmentation error.
if you run Octodash from the desktop by selecting Accessories -> Octodash it runs and works perfectly.

NOTE:  the latest version of raspian has removed the autologin feature of raspi-config so that part cannot be done, however the unit does boot all the way to desktop without any login and as stated Octodash runs fine if you launch it from desktop.

The desired result for me is to somehow launch this automatically either from console if possible or by somehow auto running from the desktop.

cheers
NicW

nicwilson58, 2020-11-27:
[UPDATE]
I am still interested in your comments however I seemed to have resolved this.   from the install at the point it was above I ran the adafruit installer described on this website then installed the 3.5" resistive touchscreen option,  following that I re-ran the "sudo ./LCD35-show" command and then restarted.  Octoprint now runs on boot before the desktop shows and works fine.   Such a complex OS that is so user non friendly, but eventually got it working.

thegarbz, 2020-11-28:
Good to hear you got it working, but you touched on the real problem with Linux, with great customisability comes great frustration. No two systems are alike. Running this on Raspbian instead of Octopi means you're going to have to tinker.

UnchartedBull, 2020-11-30:
If you installed Raspbian with Desktop you should add OctoDash to the Raspbian startup programs. The normal Raspbian Desktop will boot itself before the Window Manager that OctoDash uses by default, that's why you haven't seen OctoDash. The normal installer for OctoDash is designed for a console only distro, that's why it might be causing trouble on systems with a GUI already on them.

closed 2020-11-30 by UnchartedBull

nicwilson58, 2020-12-01:
> If you installed Raspbian with Desktop you should add OctoDash to the Raspbian startup programs. The normal Raspbian Desktop will boot itself before the Window Manager that OctoDash uses by default, [...]

I did install Desktop,  when it did not work,  your troubleshooting steps did the desktop install but it still did not work after than with no changes to desktop I resolved it as stated.