UnchartedBull/OctoDash #1217: OctoDash not starting automatically

gone-bush, 2020-11-27:
### What is happening:

The system boots normally, the screen is initialised and _ratposion_ starts - it's banner is displayed on the top right of the screen - but OctoDash does not start. When I log on to the system via ssh and execute

`ps -eF | grep -i octodash`

it shows that OctoDash is not running.

### What I've tried:

MANY things, including
```
export DISPLAY=:0
octodash

```
that works.

### The files:
(Yes, I know that they are not "standard" - but the standard files don't work, but neither do these :( )

**$HOME/.bashrc**
```
...
date >>x_out.txt
echo before test >>x_out.txt
echo SSH_CLIENT=$SSH_CLIENT= >>x_out.txt
echo SSH_TTY=$SSH_TTY= >>x_out.txt

if [ -z "$SSH_CLIENT" ] || [ -z "$SSH_TTY" ]; then
	export DISPLAY=:0
	echo before xinit >>x_out.txt
	xinit -- -nocursor
	echo after xinit >>x_out.txt
fi
```
**$HOME/.xinit**
```
#!/bin/bash

xset s off
xset s noblank
xset -dpms
```
**$HOME/x_out.txt**
```
Sat 28 Nov 09:15:48 AEST 2020
before test
SSH_CLIENT==
SSH_TTY==
before xinit
after xinit
Sat Nov 28 09:16:36 AEST 2020
before test
SSH_CLIENT=172.16.0.20 35278 22=
SSH_TTY=/dev/pts/0=
```

thegarbz, 2020-11-28:
Check the file is called .xinitrc and not .xinit like you wrote.
Also ... is that it? Or are you not actually starting anything other than X? You're missing the last two lines:

  ```
  #!/bin/sh

    xset s off
    xset s noblank
    xset -dpms

    ratpoison&
    octodash
```

gone-bush, 2020-11-28:
Oops, it is actually .xinitrc - sorry.

And the "..." denotes lines I have omitted for brevity and clarity

thegarbz, 2020-11-29:
So where do you start ratpoison? Is it not in xinitrc? 
When trying to find out what doesn't start it's always best to look at the last thing which did start. So how does ratpoison start? Is it by a script? Does the script block at that point (i.e. is the ampersand after ratposion and are you trying to start octodash in the same place?)

Show us the files which you're actually using to start ratpoison and try to start octodash.

gone-bush, 2020-11-29:
Because nothing was happening, I started to experiment. The files I have posted are the ones currently in use.

From my testing, I have assumed that ratpoison is started from the code in .bashrc, hence the "echo's" and to remove any confusion I removed those lines from .xinitrc.

Having those lines in .xinitrc and not has not changed anything.

thegarbz, 2020-11-29:
Again where is ratpoison currently started? You said ratpoison starts right? How? Post that file in full and it would give us a starting point since we know at least that file is working.

gone-bush, 2020-11-29:
SOLVED!!

Because I am not using OctiPi - I manually installed OctiPrint - I originally tried to also install OctoDash manually. That failed for reasons I cannot recall. I then used the automated install script.

The problem was that that there was the systemd file for display-manager.service which obviously was fighting with the ratpoison in .xinitrc.

I move display-manager.service aside and all is well.

Thank you very much for your assistance - very much appreciated.

thegarbz, 2020-11-29:
No problem :)

closed 2020-11-29 by thegarbz